zotil/TorBox #144: Torbox without password?

vanjavk, 2022-01-28:
if I set password to empty it just doesn't do anything.

nyxnor, 2022-01-29:
which password? User password for SSH? Lan Passowrd? Tor Control password?

Please describe what is `just doesn't do anything.`

Anyway, I probably think doing anything is the expected behavior as you should have a passowrd for security reasons and I think that of the aforementioned passwords class, they all check if password is not empty to set a new password, so leaving it blank is not gonna work.

Use a strong and secure password, use a password manager such as [keepassxc](https://keepassxc.org/) to help on your journey.

vanjavk, 2022-01-29:
Sorry, setting WLAN password to nothing/empty doesn't do anything, I'd want to have no WLAN password. KeePass is great!

nyxnor, 2022-01-29:
Here is the code https://github.com/radio24/TorBox/blob/2b27450e2ca3bad5fc50d4e028c41b46fd76078b/menu-config#L233

The maintainer radio24 will respond about letting the password be empty or not.

radio24, 2022-01-29:
Hello all
Sorry for my late reaction; currently, I'm sick in bed. This is one of these requests, which we should add to a "toxic" menu that is only available in an advanced mode. Usually, people want to have a password and removing the check is dangerous because if someone does something wrong, he could unintentionally remove the password.

@vanjavk are you able to change the code on your TorBox by yourself?

vanjavk, 2022-01-29:
Hey, yeah. I've edited /etc/hostapd/hostapd.conf

commented 3 lines and changed wpa to 0. Now it works without password.

```
wpa=0
#wpa_passphrase=torbox123
#wpa_key_mgmt=WPA-PSK
#rsn_pairwise=CCMP
```

on unrelated note, big download button leads to old version of torbox download on frontpage: 

`<a href="https://www.torbox.ch/data/torbox-20210802-v042.gz"><img loading="eager" class="alignright size-medium wp-image-133 jetpack-lazy-image jetpack-lazy-image--handled" src="https://www.torbox.ch/wp-content/uploads/2018/01/download_button-300x74.png" ...`

radio24, 2022-01-29:
Thank you for the hint with the wrong download path! It is now fixed.