zotil/TorBox #146: Onion Service name bug

Zotil, 2022-01-30:
Testing TFS live with the TorBox I created an Onion Service with a name that includes a / and $, it failed and Tor service went down.

Logs:
```
torbox@TorBox050:~ $ sudo tail /var/log/tor/notices.log
Jan 30 20:39:02.934 [notice] Opened Transparent pf/netfilter listener connection (ready) on 192.168.42.1:9040
Jan 30 20:39:02.934 [notice] Opening Transparent pf/netfilter listener on 192.168.43.1:9040
Jan 30 20:39:02.934 [notice] Opened Transparent pf/netfilter listener connection (ready) on 192.168.43.1:9040
Jan 30 20:39:02.934 [notice] Opening Control listener on 127.0.0.1:9051
Jan 30 20:39:02.934 [notice] Opened Control listener connection (ready) on 127.0.0.1:9051
Jan 30 20:39:02.000 [notice] Not disabling debugger attaching for unprivileged users.
Jan 30 20:39:02.000 [warn] Error creating directory /var/lib/tor/services/test/i$ng: No such file or directory
Jan 30 20:39:02.000 [warn] Error loading rendezvous service keys
Jan 30 20:39:02.000 [err] set_options(): Bug: Acting on config options left us in a broken state. Dying. (on Tor 0.4.6.9 55a55fbdda1adbef)
Jan 30 20:39:02.000 [err] Reading config failed--see warnings above.
```

After deleting the HiddenService with the bad name from /etc/tor/torrc, I restarted Tor with success.

nyxnor, 2022-01-30:
I will block special characters for the service name. Filtering would not be ok because that create a bad habit and the name at the end won't be as expected.

I will make it accept only `[a-zA-Z0-9]` and `_`, `.`.

nyxnor, 2022-01-30:
a programmer is finishing his beer pub software and to be sure it won't break, he tests it:
- asks for 1 beer
- asks for 2 beers
- asks for 100 beers
- asks for -15 beers
- asks for 0 beers

It doesn't fail, so he releases it.

The first client asks for a beer with tequila, the pub explodes.

nyxnor, 2022-01-30:
Did not found a way to block `$`, it deletes everything after it and the service will be created either way. `$` symbolizes end of line, parameter.

```
$ ./bin/onionjuggler-cli on -s asdasdsad -S unix -p 80 --getcliopt
asdasdsad
$ ./bin/onionjuggler-cli on -s as$dasdsad -S unix -p 80 --getcliopt
as
```

The rest is blocked.

closed 2022-07-01 by radio24

nyxnor, 2022-08-11:
Other special characters to block is on client names, they should follow the same limitations as server names.

nyxnor, 2022-08-24:
https://github.com/nyxnor/onionjuggler/blob/5a013436e0c76c4d51de954e56512e5fe83ec39a/usr/share/onionjuggler/defaults.sh#L223

Block special chars, but allow names to have `.`, `-`, and `_`.
- block dot `.` from the beginning because that is for hidden files.
- block hifen/dash `-` from the beginning because that is for --options.