zotil/TorBox #151: VoIp doesn't work on mobile phones

Nonie689, 2022-03-20:
WhatsApp and Satelite VoIP doesn't work.

The ringing is working. But if the calling is accepted no voice are transferred.

nyxnor, 2022-03-25:
Can you force `TCP only mode` with any of these apps?

radio24, 2022-03-25:
Sorry for my absence and the late answer.
Tor doesn't support UDP, which VoIP usually uses. As nyxnor has suggested, a workaround could be to force an application to use TCP, which is possible with some VPN clients. However, WhatsApp doesn't let you force VoIP through TCP to do my knowledge. In other words: as long as Tor doesn't support UDP, it will not work through TorBox.

nyxnor, 2022-03-26:
It is possible to use mumble, anyone can host a mumble server, even as an onion service.
Anyway, it is server based, not phone number based. Clients can be android (mumla), ios (idk) or desktop (mumble)

Nonie689, 2022-03-27:
> App doesn't let you force VoIP through TCP to do my knowledge

If I use orbot, the calling works.

Have you any idea why it is possible to use voip with orbot?

nyxnor, 2022-03-27:
Post deleted by author, id https://github.com/radio24/TorBox/issues/151#issuecomment-1079838166
>> App doesn't let you force VoIP through TCP to do my knowledge
> If I use orbot, the calling works.
> 
> Have you any idea why it is possible to use voip with orbot?

Orbot acts as a VPN and tunnels udp over Tor, https://github.com/guardianproject/orbot/commit/2d98a29d821f563f262e2efb71d47ecc9e88a823
Don't know how to do this for TorBox, some [documentation](http://www.dds6qkxpwdeubwucdiaord2xgbbeyds25rbsgr73tbfpqpt4a6vjwsyd.onion/wiki/Tunnel_UDP_over_Tor) for the future.

radio24, 2022-04-01:
In theory, it is simple: start a VPN client with TCP support on the TorBox client ([VPN over Tor](https://www.torbox.ch/?page_id=112)). However, this is coming with a catch: this cancels out any security and anonymity advantages Tor offers. Technically, with this method, the VPN provider sees all your data traffic. You have to trust the VPN provider so much that you might not even use Tor at all.

closed 2022-07-01 by radio24