zotil/TorBox #154: [BUG] How to select the correct wlan interface?

Nonie689, 2022-06-21:
I want to select the correct wlan interface but it changes sometimes the interface name from wlan0 to wlan1

and I want to have have a second torbox gateway on lan only mode on the same torbox rapsberry.

hope you can help me.

PS: I have wrote you an email with other question.

Nonie689, 2022-06-21:
PPS:

The lan and wlan1 gateway and the rpi lan - needs to be isolated by iptables/nftables rules.

Not sure that it is on default.

Nonie689, 2022-06-21:
I am not sure.

I believe to add a virtual gateway on lan0 is:

modprobe dummy

    ip link add veth0 type dummy  &> /dev/null
    ip addr add 10.0.10.1/24 brd + dev veth0 label veth0:0  &> /dev/null
    ip link set dev veth0 up  &> /dev/null

and to activate it as extra local lan gateway wiht Isolation by iptables/nftables rules I do not know yet.

Nonie689, 2022-06-21:
PS: A better name as virtual network interface would be vgateway0 (not veth0)

radio24, 2022-07-01:
I had the problem with switching wlan0 with wlan1 only once with a [Raspberry Pi 3 Model B](https://www.raspberrypi.org/products/raspberry-pi-3-model-b/). That would be my first question: what kind of hardware did you use when you had this wlan-switching.

Regarding the second point: I'm not sure what you exactly want to do. All possible combinations of connections can be found in the documentation (see [here under the entry "Menu entries 5-10"](https://www.torbox.ch/?page_id=775)).

Other than mentioned in the title of this issue, even if your hardware is buggy, I don't see a bug in TorBox. I will rename the title to prevent any confusion.

closed 2022-07-11 by radio24