zotil/TorBox #184: Wifi access point does not work after reboot.

DEC-entralized, 2022-11-26:
Hello the TorBox wifi access point sometimes fails when I shut down the TorBox and then turn it back on. Or even after a reboot.
There is then no TorBox wifi access point and the error remains even after switching on and off several times.
I have set up a TorBox with the installation script and Raspberry Pi OS Lite 64bit.
So I had access to the TorBox with a screen and not only via ssh.
The wifi also fails after reboots. But the box boots normally.
What could be the reason for this?
I have changed the SSID and the Wlan password. Can this produce errors?

DEC-entralized, 2022-11-26:
So I found out that the entry number 10 in the configuration menu is always deactivated when I restart the TorBox or turn it off and on again.
I can manually re-enable item 10 in the configuration menu, but it's a pain to have to do it every time.
In addition, the selection for the Internet source in the main menu of the TorBox, i.e. whether the TorBox gets its Internet via Wlan or LAN etc. is always deactivated after a restart. I have to correct this manually every time.

radio24, 2022-11-26:
Hello 
Thank you for your feedback. I assume you use TorBox v.0.5.1 (master branch) on a Raspberry Pi 4. 
So far, I have never had the problem that if TorBox's wifi initially worked, it suddenly stopped working after a reboot, even with a changed SSID and password. Did you do something else which was wifi related? For me, that information is important to isolate the error. Here is a list of possibilities:
- used main menu entry 7 to switch the wifi to a USB wifi controller instead of the internal chip;
- used in the config sub-menu entry 5-7 to change the regulatory domain, the GHz band or/and the channel;
- used in the config sub-menu entry 9/10 to hide/disable TorBox's wifi.

What do you mean by "entry number 10 in the configuration menu is always deactivated"? Initially, when TorBox's wifi works, entry number 10 should read as follows: "Disable TorBox's WLAN". It means if you press enter on this menu entry, it will disable TorBox's Wifi. Did you use it? Did you choose "temporary" or "permanently" in the following dialog? How does this menu entry read after a reboot?

Do you see any error messages at the end of the boot process after rebooting TorBox? Could you please attach a screenshot of these messages? For me, also good to know is the following points (after a reboot with failed wifi):
- the content of `/etc/hostapd/hostapd.conf`
- what is the output of `sudo systemctl status hostapd.service`?
- what is the output of `sudo systemctl status rc-local.service`?

DEC-entralized, 2022-11-26:
Did you do something else which was wifi related? For me, that information is important to isolate the error. Here is a list of possibilities:

used main menu entry 7 to switch the wifi to a USB wifi controller instead of the internal chip;
no I have not changed anything in the point 7.
I had defined the internet source as ethernet cable (ETH0) in the initial dialog.

used in the config sub-menu entry 5-7 to change the regulatory domain, the GHz band or/and the channel;
Yes in the config-sub-menu I have set the WiFi regulators to DE.
Additionally I changed the frequency to 5GhZ channel 40 with 80MhZ.

used in the config sub-menu entry 9/10 to hide/disable TorBox's wifi.
No in the config-sub-menu point 9 and 10 I have not changed anything.
At point 9 it says:
Hide TorBox'es Wifi
However at point 10 it always says Activate TorBox'es Wifi instead of Deactivate TorBox'es Wifi after a reboot.

What do you mean by "entry number 10 in the configuration menu is always deactivated"? Initially, when TorBox's wifi works, entry number 10 should read as follows: "Disable TorBox's WLAN". It means if you press enter on this menu entry, it will disable TorBox's Wifi. Did you use it? Did you choose "temporary" or "permanently" in the following dialog? How does this menu entry read after a reboot?
As mentioned above, point 10 always says Activate TorBox'es Wifi instead of Deactivate TorBox'es Wifi after a reboot.
I don't get a dialog when I want to activate it again. I press enter and it just reactivates.

Do you see any error messages at the end of the boot process after rebooting TorBox? Could you please attach a screenshot of these messages? For me, also good to know is the following points (after a reboot with failed wifi):

the content of /etc/hostapd/hostapd.conf
what is the output of sudo systemctl status hostapd.service?
what is the output of sudo systemctl status rc-local.service?

I will read these parameters this evening as soon as I get home. How exactly do I get the contents of this file?
the content of /etc/hostapd/hostapd.conf

Sorry for noob I have never done this before

DEC-entralized, 2022-11-26:
Ps. Yes I am using TorBox v.0.5.1 (master branch) on a Raspberry Pi 4.
I had updated the firmware on this PI with the update and maintance menu under point 3 but I read that this firmware is still in beta stage.
Possibly this is the reason?

DEC-entralized, 2022-11-26:
Hey I have all the information you needed :)
[image]
[image]
[image]
[image]
[image]

radio24, 2022-11-26:
First quick analysis:
- `hostapd.service`, responsible for the wifi, couldn't be started
- `rc-local.service` couldn't be executed properly

The last comment line in the `hostapd.conf` reads that 5 GHz 80 MHz settings will most likely crash `hostapd` if the country code is changed. Of course, in the `hostapd.conf`, the country code is still set to `US`, but `sudo iw reg set DE` was executed and `REGDOMAIN=DE` set in `/etc/default/crda`. It may lead to problems.

Please, try one of the following possibilities:
1. Change to 5 GHz with a max frequency of 40 MHz.
2. Set the wifi regulators to Global or US and the wifi to 5 GHz with a frequency of 80 MHz.

Variant two is more likely to fail. From what I have seen, using 80 MHz can be problematic.

Probably the problem with `rc-local` is tied to `hostapd`. However, I fixed something last month (see [here](https://github.com/radio24/TorBox/commit/d85c0ba83aff619f795a930e105b85b2ccc9eb70)). To be sure that you have the latest `rc.local` running, please do the following:
- Update the TorBox menu with entry 5 in the Update and Maintenance submenu
- Go to the command line and execute the following command: `sudo cp etc/rc.local /etc

If the error persists, please do the following:
- Go to the Configuration submenu and set logging to high (entry 16)
- Restart TorBox, go immediately to the command line, execute `sudo systemctl status rc-local.service` and attach the output for further investigation.

DEC-entralized, 2022-11-27:
Please, try one of the following possibilities:

Change to 5 GHz with a max frequency of 40 MHz.
Set the wifi regulators to US and the wifi to 5 GHz with a frequency of 80 MHz.

Hi I have limited the frequency to 40MhZ and also set the regulations back to US so default.
After reboot wifi was available again.
But still the config-menu said enable TorBox's wifi instead of disable.

Then I performed these steps:

Update the TorBox menu with entry 5 in the Update and Maintenance submenu
Go to the command line and execute the following command: `sudo cp etc/rc.local /etc
If the error persists, please do the following:

Go to the Configuration submenu and set logging to high (entry 16)
Restart TorBox, go immediately to the command line, execute sudo systemctl status rc-local.service and attach the output for further investigation.

The output of the command was:
[image]

After the updates and a reboot it said disable again and not enable in the config menu under 10

BUT one thing ist still not okay.
In the main menu under point 5 there is no arrow after a reboot.
And Tor can not bootstrap either.
If I activate it by hand again Tor bootstrap also again to 100%.
[image]

radio24, 2022-11-27:
Thank you for you patience. Let's summarize the findings.

The **problem with the wifi** seems to be connected with the frequency and/or the change of the regulatory domain. Also, the **problem with the rc-local.service** seems to be fixed with the updated `rc.local`.

Regarding the arrow: that's possible, but has no influence on the functionality. It seems that the `eth0` interface of the Raspberry Pi didn't automatically get an IP address or a route. With using the main menu, TorBox will check in the background some network parameters, and if it seems that there is no IP, it will try to request one. If you interested to try it out manually, the arrow is connected to the output of the following command:
`sudo timeout 5 sudo route | grep -m 1 default | tr -s " " | cut -d " " -f8`

The result of the command should be the Internet interface (for example, `eth0`).

By the way, if you interested to test the next version of TorBox, please check our discussion page, named "T[orBox v.0.5.2 beta - contribute to our development](https://github.com/radio24/TorBox/discussions/182)"

DEC-entralized, 2022-11-27:
Hello again I tried again with another PI 4 and set up a completely new TorBox.
Unfortunately exactly the same thing happens.
I just filled in the initial dialog as it was already set. I have selected Ethernetcable (ETH0) as internet source.
Then I changed the admin password and the wifi password and nothing else. After that I did a reboot and the wifi was gone again.
I have then accessed through a directly connected screen the logging set to high and here again read out all the info.
So as I said I have changed nothing at all except the wifi and admin password.
I have tried it on 2 different PI 4. On both this happens.
No matter whether I set up via image or installscript.
[image]
[image]
[image]

radio24, 2022-11-27:
What does `journalctl --xe` and `journalctl --no-page -u hostapd.service` gives you?
Could you try to start `hostapd` manually with `sudo systemctl start hostapd.service`?
Did you update and replace `rc.local` with `sudo cp etc/rc.local /etc`?

DEC-entralized, 2022-11-27:
What does journalctl --xe and journalctl --no-page -u hostapd.service gives you?
[image]
[image]

Could you try to start hostapd manually with sudo systemctl start hostapd.service?
[image]

Did you update and replace rc.local with sudo cp etc/rc.local /etc?
yes

radio24, 2022-11-27:
See screenshots 2 and 3: The wifi passphrase is too short. It must have a length of at least 8. However, it is also good to know for me because I have to add a check in the Configuration sub-menu.

DEC-entralized, 2022-11-27:
okay yes i took a 8 digit one and now the wifi works again. some special characters seem to cause problems too. 
sorry for the trouble but at least I could help you to the idea to build a check :P

closed 2022-11-27 by radio24