zotil/TorBox #186: Raspberry Pi 3B v1.2 no SSH connection

CrashedCrash, 2022-12-18:
Salute,
as described in the title, I just can't get an SSH connection after booting the TorBox....

No matter if via 192.168.42.1 or IP set in the router.

With the image that can be flashed, the WLAN "TorBox051" appears and I can connect but no SSH connection can be established....

Have also tried the install scripts based on Debian or Ubuntu. Where the installation works without errors. But after reboot no wlan appears

If I install again a "normal" OS, no matter which, I can again establish an SSH connection.

Under the Raspberry Pi 4B works, everything as it should, but this I need for something else.

Maybe someone can tell me how to get the debugs to post here?

CrashedCrash, 2022-12-24:
Soo sorry for the long wait time. With an Display attached I was able to configure the TorBox.
And there shows everything fine.. "TOR IS RUNNING" and the graph shows activity.

Now i`m also able to connect to the AP.
But I cant go with it to the WWW.. DNS problems and sometimes just hanging and loading before he says me no connections.

Looks like I need to buy an Raspberry 4, there seems to work everything

CrashedCrash, 2022-12-24:
Also at boot the raise network interface failed to start

CrashedCrash, 2022-12-24:
[image]

It seems like he have a connection but dont share it with the clients

CrashedCrash, 2022-12-24:
ssh still dont work

radio24, 2022-12-30:
Hello again

Regarding the "raise network interface failed to start" message: it does not automatically mean something is wrong.

I guess there is no problem with the AP anymore. However, not being able to log in with ssh is strange. To search for the reason, you can try the following:
- Check on the client, if it gets the IP address. Not static IP address, but full DHCP must be activated on the client to get the correct IP, Gateway and DNS address. Most likely, that's not the case because that would explain the problem with the SSH connection and the DNS problems.
- On the terminal, you can check if the ssh service is running with `sudo systemctl status sshd.service`
- Check if the `wlan0` interface has the correct IP address with `ifconfig`
- Re-run the first start-up dialogue by choosing entry #11 in the "Update and Maintenance" sub-menu. In the process, activate TorBox's automatic counter actions, but deactivate Vanguards. For more information about the first start-up dialogue, see **[here](https://www.torbox.ch/?page_id=2637)**.
- Share more information to analyze the problem by [sending me](mailto:anonym@torbox.ch) the files listed [here](https://www.torbox.ch/?page_id=112#i-cant-get-torbox-to-work-and-need-more-detailed-help-what-should-i-do).

The yellow log entries on the top of the statistics monitor could be problematic. You are connected with a entry guard which is failing. With TorBox's automatic counter actions activated (see above), TorBox will automatically try to fix the problem and choose another entry guard. A failing entry guard can also lead to DNS resolution and connection problems, but it doesn't explain your problem connecting TorBox with ssh.

Last but not least, in the next 7-10 days, TorBox v.0.5.2 should be released, which has numerous fixes and lets you filter out slow relays. That was responsible for terrible slow connections in the last months.

radio24, 2023-01-24:
Issue closed due to inactivity

closed 2023-01-24 by radio24