zotil/TorBox #189: incorrect CHANGE-IT for first connect wifi ?!

sajjads24, 2023-01-06:
hi
i am install torbox 5.2 on raspberry pi 3 but for first connect with wifi incorrect defualt password : CHANGE-IT
now how fix it ?
and set new password for wifi ?

sajjads24, 2023-01-06:
i am manual change file hostapd.conf

`# This configuration file was modified by TorBox
# Please don't change it manually, unless you know what you are doing
# If you delete values (even the ones with #), the TorBox menu will not re-add them again!
# You find a backup of the default file in torbox/etc/hostapd
# IMPORTANT: don't remove or change the "country_code=US" value! Otherwise, the
# 5 GHz 80 MHz settings will most likely crash hostapd!

interface=wlan0
driver=nl80211
ssid=TorBox
country_code=US
hw_mode=g
channel=6
ieee80211n=1
ieee80211ac=1
wmm_enabled=1
#ht_capab=[HT40-][HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]
#vht_oper_chwidth=1
#vht_oper_centr_freq_seg0_idx=42
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=18121812
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
`
but can not connect wifi with mobile samsung and pc !
how fix and connect wifi ?

radio24, 2023-01-06:
The image file contains [this](https://raw.githubusercontent.com/radio24/TorBox/master/etc/hostapd/hostapd.conf) configuration. Thus, I don't think it is a configuration error and, therefore, too little information to help you.

- Did you use the image file or try to update an older system?
- Do you see any error messages at the end of the boot process after rebooting TorBox? Could you please attach a screenshot of these messages? 

For me, also good to know is the following points (after a reboot with failed wifi):
- the content of /etc/hostapd/hostapd.conf
- the content of run/torbox.run
- what is the output of sudo systemctl status hostapd.service?
- what is the output of sudo systemctl status rc-local.service?

In the meantime, I will set up a test system using the new image file to see if there is an error. Honestly, because I already tested it, it is doubtful that something on the image is corrupt. Nevertheless, one never knows...

radio24, 2023-01-07:
I just uploaded [the fixed image using the same URL](https://www.torbox.ch/data/torbox-20230102-v052.gz) and [updated the SHA256 hash](https://www.torbox.ch/?page_id=1128). Feedback would be greatly appreciated.

closed 2023-01-24 by radio24