zotil/TorBox #209: Building from scratch - No wheels for opencv-python-headless

airgap0, 2023-03-09:
Trying to build (the most useful) Torbox from scratch: Starting with a Pi 3 and going down to a Raspberry Pi Model B Plus Rev 1.2 I always end up with an error when it comes to `sudo pip3 install opencv-python-headless`. First of all the process runs virtually for hours even on a Pi3. In the end there is an error message like "could not make wheels for opencv.python-headless" or similar wording. 
Now that is what I did:

- Using Pi Imager to burn a 32bit raspberryOS lite on a 16GB SD. Via configuration user is set to torbox and hostname is also torbox. SSH is enabled, user PW ist set to ???
- The Pi is connected to the LAN over ethernet
- From some Laptop I ssh into the pi
- sudo raspi-config > WiFi Country set to AT, no autologin
- Starting with `sudo printf "nameserver 1.1.1.1\nnameserver 1.0.0.1\nnameserver 8.8.8.8\nnameserver 8.8.4.4\n" | sudo tee /etc/resolv.conf` everything ist entered  as per your [instruction](https://www.torbox.ch/?page_id=205)

Everything is fine up to `sudo pip install opencv-python-headless `

Please tell me, if you need further information (and where to look for it)

Regards

zotil, 2023-03-09:
> First of all the process runs virtually for hours even on a Pi3. In the end there is an error message like "could not make wheels for opencv.python-headless" or similar wording.

Maybe a Raspberry Pi 1 will be a bit slow to build these libraries.
Regarding the error message, try upgrading pip before running the installs: `sudo pip3 install --upgrade pip`

In case you want to test another alternative, you can install opencv with apt: `sudo apt install python3-opencv`

airgap0, 2023-03-10:
This is the output when `sudo pip3 install opencv-python-headless` died.
Remark: I followed zotil's suggestion and upgraded pip, alas to no avail.

      [100%] Linking CXX static library ../../lib/libopencv_gapi.a
      [100%] Built target opencv_gapi
      Install the project...
      -- Install configuration: "Release"
      -- Installing: /tmp/pip-install-305og5q2/opencv-python-headless_7654f9d3a3                                                                                                                                   1945fd9aa59463b3ab01de/_skbuild/linux-armv6l-3.9/cmake-install/share/licenses/op                                                                                                                                   encv4/opencl-headers-LICENSE.txt
      -- Installing: /tmp/pip-install-305og5q2/opencv-python-headless_7654f9d3a3                                                                                                                                   1945fd9aa59463b3ab01de/_skbuild/linux-armv6l-3.9/cmake-install/lib/opencv4/3rdpa                                                                                                                                   rty/libade.a
      -- Installing: /tmp/pip-install-305og5q2/opencv-python-headless_7654f9d3a3                                                                                                                                   1945fd9aa59463b3ab01de/_skbuild/linux-armv6l-3.9/cmake-install/share/licenses/op                                                                                                                                   encv4/ade-LICENSE
      -- Installing: /tmp/pip-install-305og5q2/opencv-python-headless_7654f9d3a3                                                                                                                                   1945fd9aa59463b3ab01de/_skbuild/linux-armv6l-3.9/cmake-install/include/opencv4/o                                                                                                                                   pencv2/cvconfig.h
      -- Installing: /tmp/pip-install-305og5q2/opencv-python-headless_7654f9d3a3                                                                                                                                   1945fd9aa59463b3ab01de/_skbuild/linux-armv6l-3.9/cmake-install/include/opencv4/o                                                                                                                                   pencv2/opencv_modules.hpp
      -- Installing: /tmp/pip-install-305og5q2/opencv-python-headless_7654f9d3a3                                                                                                                                   1945fd9aa59463b3ab01de/_skbuild/linux-armv6l-3.9/cmake-install/lib/cmake/opencv4                                                                                                                                   /OpenCVModules.cmake
      -- Installing: /tmp/pip-install-305og5q2/opencv-python-headless_7654f9d3a3                                                                                                                                   1945fd9aa59463b3ab01de/_skbuild/linux-armv6l-3.9/cmake-install/lib/cmake/opencv4                                                                                                                                   /OpenCVModules-release.cmake
      -- Installing: /tmp/pip-install-305og5q2/opencv-python-headless_7654f9d3a3                                                                                                                                   1945fd9aa59463b3ab01de/_skbuild/linux-armv6l-3.9/cmake-install/lib/cmake/opencv4                                                                                                                                   /OpenCVConfig-version.cmake
      -- Installing: /tmp/pip-install-305og5q2/opencv-python-headless_7654f9d3a3                                                                                                                                   1945fd9aa59463b3ab01de/_skbuild/linux-armv6l-3.9/cmake-install/lib/cmake/opencv4                                                                                                                                   /OpenCVConfig.cmake
      -- Installing: /tmp/pip-install-305og5q2/opencv-python-headless_7654f9d3a3                                                                                                                                   1945fd9aa59463b3ab01de/_skbuild/linux-armv6l-3.9/cmake-install/bin/setup_vars_op                                                                                                                                   encv4.sh
      -- Installing: /tmp/pip-install-305og5q2/opencv-python-headless_7654f9d3a3                                                                                                                                   1945fd9aa59463b3ab01de/_skbuild/linux-armv6l-3.9/cmake-install/share/opencv4/val                                                                                                                                   grind.supp
      -- Installing: /tmp/pip-install-305og5q2/opencv-python-headless_7654f9d3a3                                                                                                                                   1945fd9aa59463b3ab01de/_skbuild/linux-armv6l-3.9/cmake-install/share/opencv4/val                  [... cut ...]
 note: This error originates from a subprocess, and is likely not a problem wit                                                                                                                                   h pip.
  ERROR: Failed building wheel for opencv-python-headless
Failed to build opencv-python-headless
ERROR: Could not build wheels for opencv-python-headless, which is required to i                                                                                                                                   nstall pyproject.toml-based projects
torbox@torbox:~ $
```
Unfortunately I can not understand the output, I am lacking knowledge. Maybe some expert can help me - Thanks

radio24, 2023-03-10:
Don't fiddle with pip; just run
`sudo apt install python3-opencv`
which will install the prebuilt wheel

Source: https://raspberrypi.stackexchange.com/questions/140517/cant-install-opencv-on-raspberry-pi-4

airgap0, 2023-03-11:
True, `sudo apt install python3-opencv` did the trick indeed. So it might be good to modify the [instructions for manual install/build](https://www.torbox.ch/?page_id=205) of torbox accordingly. 
Thanks for pointing me in the right direction
Regards

closed 2023-03-11 by airgap0