zotil/TorBox #213: `sshd` is enabled and running, but port 22 is not allowed

salehjg, 2023-03-18:
Hi there,
I hope you guys are having a good time.
Also, thank you for investing time in this amazing project!

I was playing around with TorBox and I noticed that SSH server was not reachable from eth0. It seemed like iptables for port 22 did not allow the incoming traffic. I manually allowed port 22 and ran `iptables-save`, it fixed it up until a system reboot. 

Even trying to expose ssh to the internet fixed the problem temporarily up to a system reboot.
Something in the startup scripts is flushing the iptables rules. Im using v052 on an RPI3B.
Could you please have a look into it or help me figure out what im missing here?

radio24, 2023-03-18:
If I understand you correctly, you want to connect the TorBox with SSH from/through the Internet?

salehjg, 2023-03-18:
no, i just want to beable to connect with a ssh client on the Lan network to torbox's ssh server inorder to configure or have control over it. The issue is that port 22 tcp is blocked due to wrong iptables rules. Something on the boot flushes the rules that i manually set to allow port 22 tcp.

salehjg, 2023-03-18:
```
sudo iptables -A INPUT -p tcp -m tcp --dport 22 -j ACCEPT
sudo iptables-save
# ssh works now
sudo reboot
#ssh now does not work, have to run these commands again!
```

radio24, 2023-03-18:
If your client is connected to `eth0` and your internet is coming, let's say, from `wlan0`, there is no need to manually reconfigure `iptables`. Just connect your cable with `eth0` and then use in the main menu entry 7. However, also here, you have to do it every time after a reboot.

If you don't need TorBox's access point at all, you could first permanently disable TorBox's WLAN in the configuration menu entry 10, and then using menu entry 7.

In your approach, the following command in `rc.local` is resetting the configuration:
`sudo /sbin/iptables-restore < /etc/iptables.ipv4.nat`

Also, you are using `iptables-save` the wrong way. Try `sudo sh -c "iptables-save > /etc/iptables.ipv4.nat"`. This will replace /etc/iptables.ipv4.nat which will be loaded during the reboot. However, if you choose a new Internet source in the main menu, `iptables.ipv4.nat` will be changed again.

closed 2023-03-18 by salehjg