zotil/TorBox #228: TorBox does not work after restart

FranklinClinton, 2023-07-04:
First of all thanks for this cool project. There’s something going on however, because whenever I restart the Pi, the following happens:
- PC connects to TorBox WiFi and gets an IP
- PC cannot access internet, cannot ping TorBox (192.168.42.1), cannot ssh to TorBox
- Now I need to connect a keyboard and a screen to the pi
- I go into the menu and choose 5 (internet via eth0)
- A bunch of stuff is going on, takes 30 seconds
- Suddenly the PC has internet (through Tor) and can ping the TorBox and also SSH into it
- Day is over, shutting the Pi down
- Next day the same story.. cannot connect, need to use direct console to set eth0 as the interface…. 

Things I did:

- I’ve taken a look at the tor log right when the system has been started and the log basically shows something like “failed to find node for path #1. Discarding this circuit”.
- I tried to simply restart tor after boot without setting the internet source. Doesn’t change a lot though, it does not show 100% bootstrapped

Is this by design? If so, can I change it? If not( what’s wrong?

radio24, 2023-07-04:
I guess `rc.local` stops in the middle because of an error. Please try the following (updating `rc.local`):
- Update the TorBox menu with entry 5 in the [Update and Maintenance sub-menu](https://www.torbox.ch/?page_id=1410).
- Quit the TorBox menu (press ESC until entering the terminal).
- Use the following command to update `rc.local`: 
- -  `sudo cp etc/rc.local /etc`
- Restart TorBox

If this doesn't work, try the following (replacing `rc.local` with version 0.5.3):
- Quit the TorBox menu (press ESC until entering the terminal).
- Use the following commands:
- - `cd`
- - `wget https://raw.githubusercontent.com/radio24/TorBox/torbox-v.0.5.3/etc/rc.local`
- - `sudo cp rc.local /etc`
- Restart TorBox

If this doesn't work, try the following (adding `dhclient eth` to `rc.local` - see also [here](https://github.com/radio24/TorBox/issues/221#issuecomment-1562328448)):
- Quit the TorBox menu (press ESC until entering the terminal).
- Use the following command to edit `rc.local`:
- - `sudo nano /etc/rc.local`
- Go to the end of the file and put **before `exit 0`** the command `dhclient eth` into the file. Save with CTRL-W and exit with CTRL-X.
- Restart TorBox

I would appreciate your feedback which of these three solutions worked for you so I can take it into account for the upcoming release.

FranklinClinton, 2023-07-05:
Thanks for the fast response! I tried the 3 options:
Option 1: Did not help
Option 2: Kinda helped, it now gets an IP right after it booted and connects to the tor network successfully. I am able to ping the torbox, accessing the web interface of torbox from the PC fails with 502 Bad Gateway. Internet access is not possible. When I go into the menu via local console it shows the arrow on the menu entry I usually use (internet over eth0). So it kinda detected that. Then I hit thst menu entry and after a couple of seconds the internet access works on the PC through Tor. 
Bad Gateway for the web interface stays the same though.
Option 3: Haven’t tried because it already gets an IP Address now and this is no longer the problem. Let me know if I should to it nonetheless.

to summarize: Option 2 enables torbox to get an IP after boot. That is good. Internet access is still not working and the web interface is broken (don’t know when the web interface broke, maybe it already broke during option 1)

radio24, 2023-07-05:
Let's first focus on the Internet access (I guess you mean Internet access by the client through TorBox).
It seems that the `iptables` configuration in `/etc/iptables.ipv4.nat` is not set yet, incorrect or not loaded through `rc.local`.
You have at least to use the main menu entry 5 once to set the right `/etc/iptables.ipv4.nat`. You can check `/etc/iptables.ipv4.nat` with `less /etc/iptables.ipv4.nat` on the terminal. It should look something like that:

FranklinClinton, 2023-07-05:
So I tried a bunch of stuff and I found the following thing quite interesting:
After a reboot the PC is not able to access the internet through the TorBox. When I execute this command:
`iptables-restore < /etc/iptables.ipv4.nat`

Internet access starts working. I will add this to rc.local, but I feel like this is a bug or something

radio24, 2023-07-05:
Thank you for your detailed information. You are on the right track: the problem lies in the `rc.local`. It also means you can forget my option 3 because the problem in Issue #221 is different.

Dealing with `rc.local` is a pain in the ass because if there is an error message during the execution of this file, the execution is stopped at this point, and the rest of the commands are not executed anymore. In your case, the error occurs most probably somewhere before the line `sudo /sbin/iptables-restore < /etc/iptables.ipv4.nat`.

With option 2 it was possible to isolate the problem. However, the `rc.local` from option 2 was written for TorBox v.0.5.3 which is a little bit different. In other words, we have to make some changes in that `rc.local`:

Old: `bash /home/torbox/torbox/bin/regset`
New: `#bash /home/torbox/torbox/bin/regset`

Old: `if grep "^TFS-" /home/torbox/torbox/run/torbox.run ; then sudo /home/torbox/torbox/bin/start_tfs initial ; fi`
New: `if grep "^TFS-" /home/torbox/torbox/run/torbox.run ; then sudo /home/torbox/torbox/run/start_tfs initial ; fi`

Old: `if grep "^TCS-" /home/torbox/torbox/run/torbox.run ; then sudo /home/torbox/torbox/bin/start_tcs initial ; fi`
New: `if grep "^TCS-" /home/torbox/torbox/run/torbox.run ; then sudo /home/torbox/torbox/run/start_tcs initial ; fi`

Also, `rc.local` is using `ipset` because of a domain exclusion feature, which I added later to TorBox v.0.5.2 (probably that wasn't a good idea and caused the problems with `rc.local` in the first place). To fix that, please install `ipset` with the following command: `sudo apt-get install ipset`.

If this will not fix the issue, try to execute the command `sudo /etc/rc.local` in the terminal and look out for error messages (or post the output here again).

FranklinClinton, 2023-07-05:
I’ve gone the way of running sudo /etc/rc.local First to see any error. Haven’t thought about that earlier. 

I commented bash /home/torbox/torbox/bin/regset out by placing a # before the line and executed the rc file again. Now without any errors!
Rebooted, but still couldn’t access internet through torbox.

Then I fixed the if statements and installed ipset. Rebooted aaaaaaand, let’s see…
It does not work.

Restarting tor doesn’t lead to 100% bootstrapped. It’s stuck on 0%. The pi has a valid IP on eth0 and is able to access the internet though. 

Running rc.local manually again. No errors. No behavior change.

Running the menu again, choosing 5 to set eth0 as source. Menu didn’t show the arrow on the side of the option.

Internet access still doesn’t work.
Man this stuff feels rough.

radio24, 2023-07-05:
Ok, that's a setback. :-( 
However, it seems to be another problem --> missing flash.

The flash is based on the following check:
`sudo timeout 5 sudo route | grep -m 1 default | tr -s " " | cut -d " " -f8`

You don't have a valid route if the router doesn't give you an IP (via DHCP). Because these IPs have time to live (TTL) setting in the DHCP server, it is possible that you have manually to release the old IP (`dhclient -r` or `dhclient -r eth`) and/or to request a new one (`dhclient eth`). This is Issue #221.

Sometimes when I do tests with a TorBox on my Internet router and restart TorBox several times, the router fails to give me a new IP which results in waiting until the TTL is over or restarting the router.

FranklinClinton, 2023-07-05:
I’ve checked that the TorBox has a valid IP. It can reach 8.8.8.8 when I am using the local console on it. That does not seem to be the problem. Unfortunately.

radio24, 2023-07-05:
What is the output of `sudo route`?
What happens, if you use `iptables-restore < /etc/iptables.ipv4.nat`?

FranklinClinton, 2023-07-05:
Let me give you the current status:
- Starting Pi
- Accessing menu via direct console 
- Tor is not working according to the menu
- Option 5 (eth0) does not show an arrow 
Quitting the menu to gather some info:
- ifconfig shows a valid IP for eth0 and also for wlan0
- route -n shows two routes, one for each subnet (eth0 and wlan0). It does not show a route for 0.0.0.0 though, but I am unsure if this is by design or if this is an error.
- iptables -L shows a lot of rules 
- ping to 8.8.8.8 fails - “Network is unreachable” - maybe because of the missing default route 
Accessing the menu again 
- Choosing option 5, yes, open access
- iptables are being set up
- Menu shows “tor is working”
- testing internet access from PC through TorBox
- No luck 
Rebooting
- Adding default route manually on torbox “ip route add default via <gateway IP reachable from eth0>”
- pinging 8.8.8.8 now works from direct console on TorBox
- Accessing menu, choosing 5
- PC can successfully access internet through TorBox!
- Restarting the TorBox results in loss of the default route and therefor requires the same steps each time

radio24, 2023-07-05:
`sudo route` should output something like that:
```
Kernel IP routing table
Destination      Gateway         Genmask             Flags Metric Ref    Use Iface
default              hostxy            0.0.0.0                 UG     0          0        0    eth0
192.168.42.0    0.0.0.0            255.255.255.0   U        0          0        0    wlan0
212.x.x.x           0.0.0.0            255.255.255.0   U        0          0        0    eth0
```
No arrow will be shown if the `default` line is missing.
Even if the default route is set manually, your Internet router should provide a route via DHCP.

Let's summarize my thoughts:
- At the beginning of the thread, the route was not an issue. We changed the `rc.local`, which doesn't affect the route. In my opinion, the route problem is unrelated to the initial issue. I guess it has to do with your router (DHCP server). If the route is missing, tor will be stuck in the low percentage.
- `webssh` should work now because the malfunction was related to a broken `rc.local`. Alternatively, Termius would be a good SSH client, too.
- The problem with the failed up-state of the TorBox after booting is related to `iptables-restore < /etc/iptables.ipv4.nat`, which was not executed because of a failing `rc.local`.

At this point, I don't know how to help you further to get your default gateway from your router.

FranklinClinton, 2023-07-06:
The missing default route is not a problem on other devices that are running windows, Ubuntu and even MacOS. Since the route was there in the beginning (otherwise I couldn’t have installed ipset), I assume it has been destroyed by our troubleshooting and tempering. 

Since we found a lot of bugs and you could implement those fixes, when would 0.5.3 be available? I could then reflash and see where we go from there. Maybe that version will work out of the box.

radio24, 2023-07-06:
I'm still fixing some bugs that pop up during the test phase. If all works according to my plans, the image should be out this weekend, latest coming Monday.

If you want to try it before, you can install it from the Github repository using this manual: https://www.torbox.ch/?page_id=1168#script

The recommended command for the script is the following (as long as the branche `torbox-v.0.5.3` is not merged into `master`): 
`run_install.sh --select-branch torbox-v.0.5.3 --step_by_step`

FranklinClinton, 2023-07-07:
Sitting tight until a new image has been released and looking forward to test it! Will report back after that.

radio24, 2023-07-08:
News to report!

As I mentioned before: Sometimes, when I do tests with a TorBox on my Internet router and have to restart TorBox several times, the router fails to give me a new IP (or default route, as I have discovered today). Because, during today's test, I was confronted with the same problem you have to deal with, I took the opportunity to investigate a possible fix in more depth.

I couldn't find why TorBox sometimes doesn't get the default route, but it also happens on other Linux systems. Research on the internet brought up several different explanations and solutions, which are also different on various distributions. Usually, that's a sign that affected people also struggle to pinpoint the source of the problem. I guess the problem has something to do with `dhcp`. Fortunately, TorBox has already implemented the solution to that problem in the menu script. I took this solution and put it into `rc.local`:

```
if ip link | grep eth0 ; then
  if grep "INTERNET_IFACE=eth0" /home/torbox/torbox/run/torbox.run ; then
    if ! sudo timeout 5 sudo route | grep -m 1 default; then
      sudo ifdown eth0 2>/dev/null
      ip addr flush dev eth0
      sudo ifup eth0 2>/dev/null
    fi
  fi
fi
```

See also [here](https://github.com/radio24/TorBox/blob/25fc7dd756cff417c86004d23cce808827dce9c8/etc/rc.local#L127).

FranklinClinton, 2023-07-09:
Well that sounds great! Can’t wait to try the new image once it has been released

radio24, 2023-07-11:
The new image is online: https://www.torbox.ch/data/torbox-20230709-v053.gz

FranklinClinton, 2023-07-12:
New image works like a charm! Great success! Thanks a lot

closed 2023-07-12 by FranklinClinton

radio24, 2023-07-12:
Thank you for your feedback and support!