zotil/TorBox #264: Torbox Raspberry Pi Zero W or 2W Dongle

gozillah, 2024-01-09:
I installed TorBox 042 (32bit) on a Raspberry Pi Zero 2W (32bit) with a USB WiFi dongle as wlan1 and a Waveshare UPS HAT as Li-ion power supply. This gave me a super-portable, tiny, standalone Tor gateway with built-in wlan0 as access point for clients and wlan1 as connection to the Internet e.g. via public WiFi. Obviously, the computing power of a Pi Zero is sufficient for TorBox, and TorBox on a Pi 5 at 64bit is a nice to have (sorry for my opinion). Another conclusion is that maintaining a 32bit version of TorBox is worth considering, because TorBox 053 (64bit) works only from Pi 3B+ upwards.

But how to make a TorBox even smaller and stealthier?

My first idea was a TorBox dongle based on the same tiny and energy-efficient Pi Zero W or 2W but without Waveshare UPS HAT, connected to and supplied by a client via USB and using Pi Zero W or 2W’s built-in wlan0 as Internet connection. This would make it possible to connect a single USB-capable client (Windows or Mac notebook, mobile, ...) via e.g. a public WiFi to the Tor network without disclosing it in the electro-magnetic domain as a TorBox access point. And it would be super-portable in the form factor of a USB stick (https://www.pi-shop.ch/pi-zero-usb-stem).

According to my preliminary tests the Pi Zero W or 2W is sufficiently powered by the 500mA available through a standard USB 2.0 client port as long as the computing load is less than number crunching.

However, the current TorBox configuration options do not support the proposed architecture. A further constraint is that TorBox requires three concurrent logical channels (client Internet traffic to TorBox, TorBox Internet access, client SSH TorBox console), whereas a bare-bone Pi Zero W or 2W offers only two physical ones (one USB, one WiFi, Bluetooth not being a considered an option here). I am not sure if both the client's Internet traffic and his SSH console can be routed in parallel over USB. And finally, I am not sufficiently familiar with the TorBox software to implement my proposal myself. 

But let me propose an implementation (you may prove me wrong or have better ideas):

1.	When configured for above use case, TorBox must not autonomously activate its own WiFi access point for SSH console access after boot, potentially exposing its function as TorBox in the vicinity’s electro-magnetic domain. This requires that there will be an alternative, persistent boot configuration for above use case, deviant from the current TorBox boot configuration (where a wlan0 access point is automatically active after boot and waiting for an SSH console login).

2.	This alternative, persistent configuration requires console access over USB after boot as e.g. detailed in https://desertbot.io/blog/headless-pi-zero-ssh-access-over-usb-windows (website as of 14 December 2019, I hope this is still valid). Windows users will have to install Putty and Bonjour. Mac users will simply have to run SSH in a terminal window. TorBox at Pi Zero W or 2W would have to have a login daemon waiting behind /dev/bus/usb/001.

3.	After login the user selects any options, then finally "7  Wireless network (through the onboard chip; wlan0)" and runs through the Internet access point dialog, selecting the access point, potentially entering a WiFi password and connecting to the Internet access point.

4.	Then the USB connection is used as tethering connection to the Internet. I do not know if it is possible to keep the SSH console USB connection open in parallel to the USB Internet tethering. If it isn’t possible, then the console connection is closed. At the next boot, the alternative, persistent setup repeats from step 2 above.

5.	It should be possible to revert the alternative, persistent configuration somehow back to the standard configuration with WiFi access to the console after boot.

If we renounce on the requirement according to step 1 above, then the implementation may be somewhat simpler: Initial setup after boot will always be via SSH console accesss to TorBox wlan0 access point (as today). This connection will have to be shut down before connecting wlan0 to the Internet, e.g. by logging out of the console. Keeping the conole open in parallel with the clients tethering access to TorBox (as potentially possible with the USB connection) will not be an option. And any paramater change after connecting to the Internet will require a reboot, and traffic volume analysis via console by the user will not be possible.

Connecting an Ethernet-capable client via Ethernet cable to TorBox is also not possible with the actual TorBox, and the solution may be of similar architecture.

That’s about as far as I can go. I would be happy if anybody knowledgeable would implement this solution (and keep the 32bit version of TorBox maintained).

gozillah, 2024-01-14:
I am one step further: I installed https://github.com/bablokb/raspi2go (see "Listing 2" in https://www.raspberry-pi-geek.de/ausgaben/rpg/2020/04/pi-zero-als-universeller-usb-stick/2/) on my Pi Zero 2W / Torbox 042. Having connected the Pi's USB port to the USB port of my MacBook I can now login over USB with "ssh torbox@TorBox042.local". Then in the main Torbox menu I selected option 7 (Internet access via the built-in wlan0) and sucessfully connected wlan0 to the external WiFi access point. In the next menu Tor access was not confirmed, but this evaluation is not absolutely trustworthy according to my earlier experience. Back in the main menu, the leftarrow to the right of option 7 confirmed the wlan0 connection. However, even after having waited for several minutes, there was no connection to the Tor network. So far I have not been sucessful applying above steps on Windows 10. Any help is appreciated.

gozillah, 2024-01-15:
Unfortunately this issue starts to be a monologue. No one has mercy. Begging seems useless. Never mind, let's move on. 

I have now progressed on Windows 10 as far as I have been with my MacBook. First I had to install a Remote NDIS (RNDIS) driver for Windows 10 to make Windows believe that my USB connection to the Pi Zero is an Ethernet connection. I found step 8 and following of this website helpful: https://www.factoryforward.com/pi-zero-w-headless-setup-windows10-rndis-driver-issue-resolved/ . Or I could simply have install the driver from https://www.factoryforward.com/wp-content/uploads/2018/01/mod-duo-rndis.zip . USB does not support the TorBox IP addressing scheme (192.168.42.1). I had to download and install Bonjour from https://support.apple.com/kb/DL999 to enable my Windows SSH utility Putty (https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html) to find the Pi Zero under the login@hostname address of "torbox@TorBox042.local". 

I am currently able to configure TorBox from both MacBook and Windows. According to my intuition I should then continue by terminating the SSH connection, followed by automatic tethering of the PC to the Pi Zero via USB, similarly to tethering a mobile to a PC for mobile data access. The Pi Zero would then route any and all USB traffic to and from Tor. Resuming TorBox configuration would require rebooting the Pi Zero. At present I do not have the faintest idea how to achieve this.

radio24, 2024-01-16:
Currently, I'm sick in my bed ... will give my take as soon as I'm up again.

radio24, 2024-01-17:
**2 - Making TorBox smaller and stealthier**
I had the same thought when I first attempted the "TorBox mini" project 2,5 years ago. I used [this](https://web.archive.org/web/20190409203920/https://medium.com/@jcolond/make-a-raspberry-pi-usb-tor-stick-2d494e7f81ea) article as a starting point. However, somehow it never worked for me. If I remember correctly, I never succeeded in seeing the  Pi Zero W USB dongle as a network device on my Mac. However, as usual, I was under time pressure and postponed the entire project ... until now. 🙄

[image]

I guess, running on a Pi Zero W/2 W, TorBox should see the Internet on `wlan0` and the client on `usb0` (this is also the way to access the TorBox menu by a web browser or a SSH client - in fact, we have **only two logical channels**, not three). To look out for the Internet on `wlan0` **is not yet supported** by TorBox. That is why it will not work in your configuration. It may be a weekend job to integrate this kind of support. I will add the necessary support if you confirm that the interface names on your tested Pi Zero W/2 W are `wlan0` and `ubs0`. However, there is another catch: in the photo above are the two solderless adapters I used to transform the Pi Zero WS to "zero dongles." Both adapters have been sold out for months, and there seems to be no solderless solution for the Pi Zero 2 W. The best solution may be to use an adapter, as seen [here](https://www.pi-shop.ch/pi-zero-usb-adaptor-white-usb-otg-host-cable). These questions must be addressed because without an easy and reliable way to get the thing working, people will not use such a device, and therefore, it wouldn't be worth investing the time to make all the changes in the code.

About your thoughts:
- Not autonomously activate its WiFi access point: possible - this is a point I still have to address with the [TorBox on a Cloud ](https://www.torbox.ch/?p=3522) configuration. A flag can handle it in the torbox.run file, which will be checked during the boot process.
- About the tethering connection: if the Pi Zero W USB dongle is treated as a network device, meaning TCP/IP is used to route the data packages from the client over the USB port to the Pi, which uses `iptables` to route the packages via `wlan0` to the tor network, then I don't see any issues. It is not different from the default configuration, and data, as well as SSH connection to the device, can be used at the same time. However, **the Pi is to be identified as a network device** - this is the crucial point ... and it cannot be complicated for the user; otherwise, nobody will use it.
- Regarding the used hardware and the type of Raspberry Pi OS: today, the base should be a **Pi Zero 2 W** because it is up to five times as fast as the original Pi Zero and supports the 64-bit version of the Raspberry Pi OS.
- About creating a pre-configured Raspberry Pi OS image for the Pi Zero 2 W: the Raspberry Pi Imager should also support this: https://www.raspberrypi.com/documentation/computers/getting-started.html#raspberry-pi-imager
- About TorBox's address scheme: to separate different kinds of "modes" but integrate them in one configuration, we have to use `192.168.45.x` for the Pi.

**Conclusion**
I'm convinced that if the plugged Pi Zero 2 W acts as a network device on a client machine, the necessary adaption of TorBox can be done on a weekend and will work. In the last few months, I did a similar task with TorBox on a cloud. To me, the following information is crucial before I invest my time in it again: the interface names on an ordinary Pi Zero 2 W dongle (`wlan0`/`usb0`?) and the easy way to build for people who will use such a device.

gozillah, 2024-01-17:
I am very happy to see this issue being pushed forward, thank you! 

I confirm that the interface names on my Pi Zero 2W are usb0 and wlan0:

torbox@TorBox042:~/torbox $ ifconfig
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        ...
usb0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
       ...
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ...

Pi Zero, Pi Zero W and Pi Zero 2W have the same Pogopin access points (Vcc, GND, D+, D-) at the same locations on their underside.

The solderless USB-A adapter is available here:
https://geekworm.com/collections/raspberry-pi/Expansion-Board+Raspberry-Pi-zero
https://www.aliexpress.com/item/1005004988260923.html
https://www.aliexpress.com/item/1005003778463582.html
However this model requires inserting the Pi Zero upside down, jacking e.g. a MacBook Air up. 

This one has the Pi Zero on the upper side, avoiding the problem:
https://www.aliexpress.com/item/1005003660807048.html
https://www.aliexpress.com/item/32846959589.html
https://www.aliexpress.com/item/32956403469.html
https://www.aliexpress.com/item/1005003049906551.html

Here is yet another solderless solution:
https://www.tindie.com/products/8086net/solderless-zero-dongle-for-raspberry-pi-zero/
https://www.tindie.com/products/8086net/solderless-zero-micro-dongle-for-raspberry-pi-zero/

For a notebook a simple Micro-USB male to USB-A male cable from the Pi Zero's Micro-USB female data port to the notebook's USB-A female connector has the same effect electrically:
https://www.aliexpress.com/item/1005005842324335.html
The white official Raspberry Pi OTG Micro-USB male to USB-A female cable would require another cable with two USB-A male connectors. 

Connecting a (up-to-date) mobile would require a USB-C male to Micro-USB male cable, e.g. this one:
https://www.aliexpress.com/item/1005005602525557.html
This Android app
https://play.google.com/store/apps/details?id=org.connectbot&hl=en&gl=US
is working 
[image]
well for TorBox SSH access.

I will be happy to test your prototype images on Pi Zero 2W with MacBook Air (Monterey), Windows 10 and Android.

This is my well-performing Pi Zero 2W with TorBox 042, WiFi dongle for wlan1 and Waveshare Li-ion supply (underneath) in a 3D-printed case with cover. Plus my Pi Zero W with the upside-down USB-A adapter:

[image]

radio24, 2024-01-18:
Thank you very much for the detailed information. I just ordered a Pi Zero 2W and will work on the project, but I must warn you that I might need some time because of my workload from other projects. 

However, first things first, with your research, preparative work, and detailed feedback, you already earned a [ProtonMail $100 Gift Card](https://shop.proton.me/collections/coming-soon/products/proton-100-gift-card-digital) from our "feedback-giveaway" program. Please send a message to the TorBox email as soon as possible, and for verification, the original date, size, and SHA256 value of the file menu in the TorBox version you use on your Pi Zero 2 W.

radio24, 2024-01-18:
What is your procedure for creating the Pi Zero W / 2 W USB-gadget?

- Creating a Raspberry Pi SD Card with configured WiFi AP and SSH server
- Booting up the Pi Zero W / 2 W with the prepared SD Card
- SSH into the Pi Zero W / 2 W 
- Download and run the raspi2go tool on the Pi Zero W / 2 W (is this tool really necessary? Doesn't `modules-load=dwc2,g_ether` be enough - see also [here](https://forums.raspberrypi.com/viewtopic.php?t=357184))?
- Access from the Mac via the USB port (without installing or configuring something on the Mac?)

gozillah, 2024-01-18:
I installed a standard Torbox 042 image on a MicroSD and booted my Zero 2W with it.

TorBox 042 in its standard configuration after installation and boot does not accept SSH through the Pi's usb0 yet. In the Internet I found a lot of websites explaining the two extensions to config.txt (dtoverlay=dwc2) and cmdline.txt (modules-load=dwc2,g_ether) in order to (allegedly) be able to SSH through the Pi's USB port, but that alone did not work in my case. 

Only after I found and installed raspi2go I could SSH into Torbox 042 without problems from my MacBook Air 2015 with MacOS 12.7.2 Monterey, Linux terminal app, "ssh torbox@TorBox042.local" (192.168.42.1 will not work). That's because MacOS supports Bonjour which - on the Mac - associates the TorBox's IP address to its hostname. May be raspi2go was somewhow cooking the Pi's IP address and hostname for Bonjour. I haven't studied yet what raspi2go actually does.

I cannot remember having installed anything particular on my MacBook for TorBox042. I think Bonjour ist just part of Monterey.

Windows does not support Bonjour by default, but it is easily installed from an Internet download. However this was not enough. Per default Windows sees the Pi as a serial device, see at the top of the device manager. In this configuration Windows does not seem to be able to handle network adressing (IP or hostname). It is therefore necessary to install an RNDIS driver which somehow bends Windows' Ethernet-centric networking view into the USB port and disconnects the serial device driver from the USB port. At that point I could access the Pi via Putty just as I did with the MacBook and the Linux terminal app. 

Android supports "Service discovery" which is Google's naming of Bonjour (Bonjour = Apple trade name). I tried several SSH apps and found the ConnectBot functional. They all can SSH-connect, but for TorBox I need a 24x80 Terminal emulation, an Esc key and some cursor keys, which ConnectBot provides. I run a Samsung S21 Ultra with Android 14.

I still haven't tried TorBox 042 on the Zero W yet. TorBox CPU load on the 2W is minimal, so a Zero W may be way enough for TorBox. 

TorBox 053 did not work on the Zero 2W: Under TorBox 053 I can connect to the Zero 2W's access point (Windows Putty torbox@192.268.42.1 or MacBook torbox@TorBox042.local) but Putty or the MacBook Linux terminal will show me an empty screen. I feel that you will have to maintain a Torbox05x 32bit image for the Zero W / 2W family.

Looking forward to your prototypes I am currently designing ever better enclosures for my Zero 2W + Waveshare UPS HAT + wlan1 dongle standalone Tor accesspoint. The tiny 1'000 mAh Li-ion battery of the Waveshare UPS HAT supplies this TorBox 042 access point for hours.

gozillah, 2024-01-18:
Perhaps a heretical thought: Why bother about 64bit? The computing load on a TorBox is so low that 32bit is way enough.

radio24, 2024-01-18:
Ok 🤔

Let me explain, what I try to do... Before I want to start any version of TorBox, I will run a plain vanilla Raspberry Pi OS (32 bit light on a Raspberry Pi Zero W). I'm using the Raspberry Pi Imager with the customisation option (see [here](https://www.raspberrypi.com/documentation/computers/getting-started.html#install-using-imager) under "OS customisation"). Then I start the Raspberry Pi Zero W up, which seems to work according to the green LED, but it doesn't create a wifi AP which I can log into.

radio24, 2024-01-19:
> I still haven't tried TorBox 042 on the Zero W yet. TorBox CPU load on the 2W is minimal so that a Zero W may be way enough for TorBox.
It may be the case if tor is freshly started. The preliminary findings of my compatibility test (see above) showed that the current tor version runs fine for two days on the Raspberry Pi 3 Model B, but then the entire system slows down and freezes as the system's resources are eaten up.

> Why bother about 64bit? The computing load on a TorBox is so low that 32bit is way enough.
First, 32-bit is marked as legacy... better to switch now. On the other spectrum, we have TorBox on a cloud. There are Cloud providers that don't offer 32-bit OS anymore. Maintaining several TorBox variants is too time-intensive for me (it is not my job; I don't even earn my money in IT) - the goal is to have one system, possibly one image, to catch the most.

gozillah, 2024-01-19:
A plain-vanilla Pi OS installation will not create an AP. You will have to configure it manually. TorBox042 contains this AP configuration already. I suggest you start from there.

"In another attempt, ...": I am not surprised. As I mentioned, the two steps in config.txt (dtoverlay=dwc2) and cmdline.txt (modules-load=dwc2,g_ether)  didn't do the trick with my MacBook either. You will have to install the magic of raspi2go on the Zero W. If you configured config.txt already, raspi2go will install another identical line at the end of config.txt and add further magic elsewhere. Check cmdline.txt whether raspi2go messed it up after your initial configuration. It is said that cmdline.txt is very syntax sensitive.

As for RNDIS in [here](https://www.circuitbasics.com/raspberry-pi-zero-ethernet-gadget/) : My Windows 10 Professional contained no RNDIS driver. Before installation of RNDIS I had to download the driver from the Internet.

gozillah, 2024-01-19:
> It may be the case if tor is freshly started. The preliminary findings of my compatibility test (see above) showed that the current tor version runs fine for two days on the Raspberry Pi 3 Model B, [...]

TorBox 

> Ok 🤔
> 
> Let me explain what I am trying to do... Before starting any TorBox version, I will run a plain vanilla Raspberry Pi OS (32-bit light on a Raspberry Pi Zero W). I'm using the Raspberry Pi Imager with the customization option (see [here](https://www.raspberrypi.com/documentation/computers/getting-started.html#install-using-imager) under "OS customization"). Then I start the Raspberry Pi Zero W up, which seems to work according to the green LED, but it doesn't create a wifi AP that I can log into.
> 
> In another attempt, I skipped the wlan part. After flashing the SD Card with Raspberry Pi OS 32-bit light (without customization), I edited the config.txt and the cmdline.txt as described [here](https://www.circuitbasics.com/raspberry-pi-zero-ethernet-gadget/). Plugging the Raspberry Pi Zero W into the MAC USB port started the Pi up. The Mac identifies the Pi as a RNDIS/Ethernet Gadget. I can even ping the Pi, but I cannot SSH into the Pi.

A plain-vanilla Pi OS installation will not create an AP. You will have to configure it manually. TorBox042 contains this AP configuration already. I suggest you start from there.

"In another attempt, ...": I am not surprised. As I mentioned, the two steps in config.txt (dtoverlay=dwc2) and cmdline.txt (modules-load=dwc2,g_ether) didn't do the trick with my MacBook either. You will have to install the magic of raspi2go on the Zero W. If you configured config.txt already, raspi2go will install another identical line at the end of config.txt and add further magic elsewhere. Check cmdline.txt whether raspi2go messed it up after your initial configuration. It is said that cmdline.txt is very syntax sensitive.

As for RNDIS in [here](https://www.circuitbasics.com/raspberry-pi-zero-ethernet-gadget/) : My Windows 10 Professional contained no RNDIS driver. Before installation of RNDIS I had to download the driver from the Internet.

radio24, 2024-01-19:
Could you refresh your browser before commenting? You are referring to an old (first) version of my comment. My last update was made 8 hours ago. There, you will see some substantial progress. 

To start with, TorBox 0.4.2 is not an option because I have to understand the fundamentals to integrate it into the latest version of TorBox, which has substantial fixes regarding older versions. With the „latest version”, I don‘t mean the image. More important are the installation scripts because with these scripts, you can also compile a 32-bit version of TorBox.

gozillah, 2024-01-19:
I deleted two of my comments. I was a bit out of sync with your comments, sorry.

gozillah, 2024-01-19:
A possible solution for Android may be "reverse USB tethering": In normal USB tethering, the mobile is the modem and the PC is the client using the mobile's Internet access via USB. Our architecture is the exact reverse: The mobile is the client and TorBox is the modem providing Internet access to the client. 

Whereas Android has a built-in standard interface for normal USB tethering, there is no such OS support for reverse USB tethering. In a Android + TorBox environment we may therefore need 

1. A reverse USB tethering Android app, implementing a relay between (a) any Android app's (e.g. browser) network access (implemented by this reverse USB tethering Android app as an Android-internal interface to which unmodified standard apps are able to connect) and (b) across USB to the reverse USB tethering server in the TorBox. 

2. A reverse USB tethering server in the TorBox.

There are a "re-Link" and a "Reverse Tethering NoRoot" app in Google Playstore, together with downloadable Windows/MacOS/Linux server applications. Using a closed-source application may not be acceptable for our purpose, but there are open-source reverse tethering apps in Github [Link](https://github.com/search?q=reverse%20tethering&type=repositories). While we could provide the server application as part of the TorBox release (i.e. invisible to the user), the user will have to install a TorBox app of some sort on his Android, and we will have to provide and maintain it for a range of Android versions. I am not yet talking about iOS (which I am not familiar with). Windows and MacOS differ in that there are easy to install standard SW packages available (RNDIS driver, Bonjour).

This is going to be complex, but I (currently) don't see an alternative among the available Android USB interfaces. I wish I was wrong. I therefore suggest that we drop Android support from our todo-list, at least for the initial release or until we discover a simpler solution. 

Apropos Bonjour: Rumors qualify it as not particularly secure. One issue seems to be that hostnames are not authenticated. I do not know whether it would be possible to inject a spoofed hostname ("TorBox.local") via the external WiFi AP such that the client may directly connect to it instead of to the TorBox entry point, or whether the TorBox entry point and the WiFi AP are sufficiently isolated from the perspective of Bonjour.

Apropos RNDIS: I read here [Link](https://wiki.gentoo.org/wiki/Android_USB_tethering) that "RNDIS linux module is being deprecated and will be removed from the Linux Kernel in favor Network Control Model (NCM), but it is used by majority of devices nowadays (2023) and up to now the only way to configure working USB tethering for old Android devices."

gozillah, 2024-01-24:
I am running TorBox042 on my dual-wlan Zero 2W without problems (apart from Torbox042 being obsolete).

Torbox053 isn't working on Zero 2W. I therefore wanted to script-install Torbox053 on 32bit Bookworm:
- I installed 32bit Pi OS light (2023-12-11-raspios-bookworm-armhf-lite.img.xz, 544.9 MB on my Mac) from [link](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-64-bit) on a MicroSD, put it into my Zero 2W, booted and connected headless via SSH and router.

I then proceeded according to [link](https://www.torbox.ch/?page_id=1168#script) and footnote 1:
- Time was ok
- Updated and upgraded Bookworm
- wget was already included
- ./run_install_on_debian.sh . [link](https://www.torbox.ch/?page_id=1168#script) says run it as root, but ./run_install_on_debian.sh welcome screen says don't run it as sudo. Root is not installed, running it as user leads to unwriteable lockfiles, so I chose sudo.
- The installation takes hours and stalls at 

> ...
Collecting h11<1,>=0.9.0
>   Downloading https://www.piwheels.org/simple/h11/h11-0.14.0-py3-none-any.whl (58 kB)
>      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 58.3/58.3 kB 492.6 kB/s eta 0:00:00
> Building wheels for collected packages: opencv-python-headless
>   Building wheel for opencv-python-headless (pyproject.toml) ... -

Here the spinner turned about once in minutes, the 0.5 GB RAM did not seem to be full, but swap was non-zero. I therefore tried the same on my newly acquired Pi 5 / 8GB. There the exact same happened, only a bit faster. It stalled at the exact same spot. 

What was wrong? How can I script-install an up-to-date Torbox053 on 32bit Bookworm?

radio24, 2024-01-25:
Let's do the next step together...

I have a Raspberry Pi Zero 2 W here. On the SD Card is a Raspberry Pi OS (Legacy, Bullseye) lite 32-bit with [these three files](https://github.com/radio24/TorBox/tree/torbox_mini/install/torbox_mini). I plug the Pi into one of my Mac's USB ports and let the Pi start up. The Mac recognizes a RNDIS/Ethernet Gadget and gives it a local IP address. I log into the Pi with `ssh torbox@torbox.local` (password: CHANGE-IT).

Next, I have to connect to my wifi. I use on the Pi the following commands:
- `sudo rfkill unblock wlan`
- `sudo systemctl stop wpa_supplicant`
- `sudo ip link set dev wlan0 up`
- Find wireless networks and note down the name (SSID): `sudo iw wlan0 scan | grep SSID:`
- Build the `wpa_supplicant.conf` (<SSID> is the name of the wireless network to connect, <wifi-password> the password of <SSID>): `sudo wpa_passphrase <SSID> <wifi-password> | sudo tee -a /etc/wpa_supplicant/wpa_supplicant.conf
- Connect to the wireless network: `sudo wpa_supplicant -B -c /etc/wpa_supplicant/wpa_supplicant.conf -i wlan0`
- If needed: start dhclient to get an IP: `dhclient wlan0`

You may get error messages by connecting to the wireless network. If you get an IP address at the end of the process, then just ignore it. In case you don't get an IP address, then stop some or all of the following services: `wpa_supplicant,` `ModemManager,` `NetworkManager,` `network.`

Now, let's install TorBox:
```
cd
wget https://raw.githubusercontent.com/radio24/TorBox/master/install/run_install.sh
chmod a+x run_install.sh  
./run_install.sh --step_by_step
```

**Some remarks on your posting above**
- You have to run `run_install.sh` on Raspberry Pi OS. `run_install_on_debian.sh` is for Debian only. On Debian, the user is logged in as `root`. Therefore, the script doesn't use `sudo` internally. Using `sudo ./run_install_on_debian.sh` on Debian will lead to error messages. In your case, you could invoke the command without error messages. However, because there are differences between `run_install.sh` and `run_install_on_debian.sh`, your system will finally not work correctly.
- I had the same problem with installing `opencv-python-headless` -- I didn't find the cause, but I fixed it. The updated script installs the package via `apt-get `instead of using `pip.` Opencv is used to solve the captcha when asking the tor bridge database for bridges. Installing the Python modules should be done in about 30 minutes.
- I guess the biggest job for the Raspberry Pi Zero 2 W is to compile tor. It takes about one hour to do that. 
- If it's of interest to you, I haven't used any additional power for the Raspberry Pi Zero 2 W. I use only the power it gets from the Mac USB port.

**References**
- [Install OpenCV on Raspberry Pi: The only guide you need](https://raspberrytips.com/install-opencv-on-raspberry-pi/)

**Next**
I will try the same again, but this time with Raspberry Pi OS (Bookworm) lite 64-bit on a Raspberry Pi Zero 2 W. If this is also successful, I will modify TorBox for the Raspberry Pi Zero 2 W.
There is also an open question: How does the RNDIS/Ethernet Gadget work under a standard Linux installation like Debian, Ubuntu, Manjaro, etc.? Do these systems automatically identify the Raspberry Pi Zero 2 W if plugged into a USB port?

**At the moment of writing, the installation is still running -- I will update the post later.**

radio24, 2024-01-26:
**Raspberry Pi OS (Bookworm) lite 64-bit is a troublemaker!**
I got into serious troubles with Raspberry Pi OS (Bookworm) lite 64-bit on a Raspberry Pi Zero 2 W. It seems it doesn't boot up correctly. After a successful boot, the file `firstrun.sh` should have been deleted, which was not the case. Ergo, it sticks or freezes somewhere. I cannot confirm it because I don't have an HDMI-mini to HDMI adapter yet. 

Interestingly, the 32-bit version seems to work fine. I guess I have to rethink my stand on the 32-bit version... Anyway, it doesn't matter because the final installation script for the Raspberry Pi Zero 2 W and TorBox should run on the 32-bit and 64-bit of the actual version of the Raspberry Pi OS.

Nevertheless, why do we see these problems when the Raspberry Pi Zero 2 W has a [quad-core 64-bit ARM Cortex-A53](https://en.wikipedia.org/wiki/ARM_Cortex-A53)? There could be two reasons - too much power consumption (unlikely because adding an external power source doesn't change the situation, or the lack of SDRAM (512 MB). Anyway, [another user experienced](https://medium.com/@techhara/stick-with-32-bit-os-for-raspberry-pi-zero-2w-d5367ce5a6d4#:~:text=Given%20that%20RPI%20Zero%202W,freeze%2Funresponsive%20problem%20at%20all!) similar problems with the 64-bit version: "Switching back to 32-bit Raspberry Pi OS for RPI Zero 2W resolved freeze and unresponsive issue I have been having with the device."

**First test with run_install_mini.sh**
The first test-run of run_install.sh and the fact that the `usb0` on the Pi can be used as an interface to the Internet (USB dongle or Tethering with Android) or to the client (RNDIS/Ethernet Gadget), I decided to create a separate installation script (`run_install_mini.sh`) for the TorBox-mini project.

Steps to test the `run_install_mini.sh`:
- Create a SD Card with Raspberry Pi OS (Bookworm) lite 32-bit.
- Copy/replace [these three files](https://github.com/radio24/TorBox/tree/torbox_mini/install/torbox_mini/bookworm/IPv6).
- Plug the Pi into one of my Mac's USB ports and let the Pi start up. The Mac recognizes a RNDIS/Ethernet Gadget and gives it a local IP address. I log into the Pi with `ssh torbox@torbox.local` (password: `CHANGE-IT`).
- Connect the Raspberry Pi Zero 2 W to the wifi (see [here](https://github.com/radio24/TorBox/issues/264#issuecomment-1909426307))
- Install TorBox with the following commands:
```
cd
wget https://raw.githubusercontent.com/radio24/TorBox/torbox_mini/install/run_install_mini.sh
chmod a+x run_install_mini.sh  
./run_install_mini.sh --select-branch torbox_mini --step_by_step
```

**Updates are following later....**

gozillah, 2024-01-26:
> Let's do it together...
> 
> I have a Raspberry Pi Zero 2 W here. On the SD Card is a Raspberry Pi OS (Legacy, Bullseye) lite 32-bit with [these three files](https://github.com/radio24/TorBox/tree/torbox_mini/install/torbox_mini/bullseye). I plug the Pi into one of my Mac's USB ports and let the Pi start up. The Mac recognizes a RNDIS/Ethernet Gadget and gives it a local IP address. I log into the Pi with `ssh torbox@torbox.local` (password: CHANGE-IT).
> 
> Next, I have to connect to my wifi. I use on the Pi the following commands:
> 
>     * `sudo rfkill unblock wlan`
> 
>     * `sudo systemctl stop wpa_supplicant`
> 
>     * `sudo systemctl stop NetworkManager`
> 
>     * `sudo ip link set dev wlan0 up`
> 
>     * Find wireless networks and note down the name (SSID): `sudo iw wlan0 scan | grep SSID:`
> 
>     * Build the `wpa_supplicant.conf` ( is the name of the wireless network to connect,  the password of ): `sudo wpa_passphrase <SSID> <wifi-password> | sudo tee -a /etc/wpa_supplicant/wpa_supplicant.conf`
> 
>     * Connect to the wireless network: `sudo wpa_supplicant -B -c /etc/wpa_supplicant/wpa_supplicant.conf -i wlan0`
> 
>     * If needed: start dhclient to get an IP: `dhclient wlan0`
> 
> 
> You may get error messages by connecting to the wireless network. If you get an IP address at the end of the process, then just ignore it.
> 
> Now, let's install TorBox:
> 
> ```
> cd
> wget https://raw.githubusercontent.com/radio24/TorBox/master/install/run_install.sh
> chmod a+x run_install.sh  
> ./run_install.sh --step_by_step
> ```
> 
> **Some remarks on your posting above**
> 
>     * You have to run `run_install.sh` on Raspberry Pi OS. `run_install_on_debian.sh` is for Debian only. On Debian, the user is logged in as `root`. Therefore, the script doesn't use `sudo` internally. Using `sudo ./run_install_on_debian.sh` on Debian will lead to error messages. In your case, you could invoke the command without error messages. However, because there are differences between `run_install.sh` and `run_install_on_debian.sh`, your system will finally not work correctly.
> 
>     * I had the same problem with installing `opencv-python-headless` -- I didn't find the cause, but I fixed it. The updated script installs the package via `apt-get `instead of using `pip.` Opencv is used to solve the captcha when asking the tor bridge database for bridges. Installing the Python modules should be done in about 30 minutes.
> 
>     * I guess the biggest job for the Raspberry Pi Zero 2 W is to compile tor. It takes about one hour to do that.
> 
>     * If it's of interest to you, I haven't used any additional power for the Raspberry Pi Zero 2 W. I use only the power it gets from the Mac USB port.
> 
> 
> **References**
> 
>     * [Install OpenCV on Raspberry Pi: The only guide you need](https://raspberrytips.com/install-opencv-on-raspberry-pi/)
> 
> 
> **Next**
> 
>     * I will try the same again, but this time with Raspberry Pi OS (Bookworm) lite 64-bit on a Raspberry Pi Zero 2 W. If this is also successful, I will modify TorBox for the Raspberry Pi Zero 2 W. **Update**: There are some hiccups with Bookworm --> see [here](https://www.zansara.dev/posts/2024-01-06-raspberrypi-headless-bookworm-wifi-config/) and [here](https://forums.raspberrypi.com/viewtopic.php?p=2147215).
> 
>     * There is also an open question: How does the RNDIS/Ethernet Gadget work under a standard Linux installation like Debian, Ubuntu, Manjaro, etc.? Do these systems automatically identify the Raspberry Pi Zero 2 W if plugged into a USB port?
> 
> 
> **Update** The installation went well, but the RNDIS/Ethernet Gadget no longer worked after restarting the Pi. That's not what we want, and we must isolate the problem.

gozillah, 2024-01-26:
I had the same trouble with 64bit TorBox053 on Zero 2W. My Waveshare "UPS HAT C" [(Link)](https://www.waveshare.com/product/raspberry-pi/hats/interface-power/ups-hat-c.htm) is said to deliver up to 1.8 amps [(Link, look for "Question: What is the output current?")](https://www.waveshare.com/wiki/UPS_HAT_(C)). The UPS HAT has an I2C interface through which I can measure the actual supply current. Torbox042 on Zero 2W has an idle power consumption of about 0.35 amps, with observed maximums of about twice this value (but I cannot measure peak glitches). 0.7 amps may be too much for MAC's USB 2.0 max. 0.5 amp. You may connect a Pi 3B power supply to the "PWR" MicroUSB of the Zero 2W.

gozillah, 2024-01-27:
In the latest Raspberry Pi Imager v1.8.5, when selecting a Pi type (..., Zero, Zero 2W, Pi 3, Pi 4, Pi 5), Bookworm is exclusively selectable for Pi 4 and Pi 5, with the 64bit version proposed and the 32bit version as an option. Bullseye 32bit is proposed for Pi 3 and lower. Bullseye 64bit is an option for Zero 2W and Pi 3 only.

radio24, 2024-01-27:
> In the latest Raspberry Pi Imager v1.8.5, when selecting a Pi type (..., Zero, Zero 2W, Pi 3, Pi 4, Pi 5), Bookworm is exclusively selectable for Pi 4 and Pi 5, with the 64bit version proposed and [...]

Raspberry Pi OS (Bookworm) lite 32-bit on a Raspberry Pi Zero 2 W works without problem.

radio24, 2024-01-28:
Please see the project's updated status [here](https://github.com/radio24/TorBox/issues/264#issuecomment-1912727821). If you follow the installation guide, you should end with a working "TorBox mini". At least, for me, it worked.

I will start below a to-do list of what has to be done for the Integration of the `torbox_mini` branch into the next version of TorBox:
- [ ] Integration of `rc.local`
- [ ] Decision on the support of `hostapd` (my idea: no support)
- [ ] Depending on the decision above: no support for additional network drivers
- [ ] Decision on what connection types in the main menu have to be supported
- [ ] Decision on building and giving out a separate image file
- [ ] Figuring out how Linux/Windows has to be configured to use a RNDIS/Ethernet Gadget in general and TorBox mini in particular
- [ ] Writing a build-manual for `torbox.ch`

**More is coming later...**

gozillah, 2024-01-29:
I just built your Bookworm 32bit Torbox-mini on my Zero 2W in about 2h30!

There were a couple of warnings flashing by. It would probably make sense to remove the "Clear" commands in the install script to be able to save the full installation minutes for later analysis.

I configured the RNDIS/Ethernet Gadget in my MacBook according to your instructions. I did this during the building process. At that time my browser on the MacBook was no longer able to access the Internet through MacBook's WiFi. I have no explanation for this. In the "Network" setup utility of the Mac both WiFi and RNDIS were marked as active (green "LED").

After completion of the buid I rebootet and it actually worked !!! Absolutely great !!!

Then I ran through the initial setup process. At the completion of the 
   [+] DONE! Checking progress - please be patient!
section (100% Done) I was not able to leave via Ctrl-C (as usual) or ESC and had to reboot again. 

Nota bene: The "100% Done" message is sometimes (but not alwys) displayed in light grey (hardly visible), also in TorBox042.

After reboot I was asked whether to update the WiFi drivers. At first sight this may not make sense in the context of a separate USB dongle image. However if a user wanted to use a special USB transmitter, e.g. with an antenna socket or with higher power, then he could implement this even with a Zero 2W and updating the drivers would make sense. This would also justify to keep the wlan1 configuration option in this image. 

After having selected the external access point and restarting Tor, I got this
   HMMM... THAT DOESN'T LOOK GOOD...
as always. Never mind, it is always fake news.

Connecting to check.torproject.org gave me the green variant of the possible answers!

I see two usecases for Torbox-mini:

1. The Zero 2W USB dongle as a super-portable, surreptitious, (relatively) easy to use Tor acces for a single user. Lacking an access point it has a relatively low electro-magnetic footprint in the close environment.

2. The Zero 2W with additional WiFi adapter and optionally self-sufficient supply (my Waveshare UPS HAT C powers such a combination for more than 2 hours). It is providing a super-portable Tor access point potentially for several concurrent users. It may be operated in a self-sufficient manner at a distance from the actual users. Due to its low cost of about USD 50 (Zero, wlan1, UPS) it may be disposable in very critical situations.

gozillah, 2024-01-29:
I successfully connected Windows 10 via TorBox mini to check.torproject.org (onyl once, see below). 
Let me remember wat I did:
- I installed the RNDIS/Ethernet driver from [link](https://modclouddownloadprod.blob.core.windows.net/shared/mod-duo-rndis.zip)
- I installed Bonjour from [link](https://support.apple.com/kb/DL999?locale=en_GB)
- I connected TorBox mini to an USB port of my Windows 10 computer, whereupon Windows detected a new, unidentified network without Internet access
- I was then able to access the Torbox mini console through Putty / ssh torbox@torbox.local and configured wlan0 to the external Internet access point
- Trying to connect to the new network opened the Windows Ethernet network settings where I observed the built-in "real" Ethernet  network of my computer as well as the new network
- Opening the new network settings I disabled DHCP and manually entered the IPV4 network params (IPV6 off)
- In this network's adapter properties I allowed others to share this "Ethernet" connection and activated all services (1700 ff, which one (port?) is really used for Internet access?)
- In the extended options I entered the network address 192.168.44.1 (you could also do it in the device manager, RNDIS/Ethernet Gadget extended driver options)

Unfortunately I was unable to repeat this success ...

gozillah, 2024-01-29:
Maybe I committed a fallacy: It is not our intention to give the Zero access to the Windows PC's Internet connection, but to give the Windows PC access to the Zero's Internet connection (= tethering via RNDIS/Ethernet).

gozillah, 2024-01-30:
I designed and printed an enclosure for the Zero 2W (without header pins) and the black USB adapter ([link](https://www.aliexpress.com/item/32956403469.html)):

[image]

[image]

[image]

It is easy to print and (re-)uses all parts of the adapter (including screws and nuts) minus the acrylic bottom plate (which gave me the z-height to re-use the screws). The MicroSD card is accessible without opening the enclosure. 

It is a bit wider than a normal USB connector, so it may obstruct nearby PC ports, but I can't help.

gozillah, 2024-02-02:
I further refined the enclosure (LED peephole, assembly cams at the corners):

[image]

The screw holes are square to make the slicer route squarely instead of round, which gives better results.

Here is the 3D-printable file in case you want to give it a try: [240202_ZeroUSB.zip](https://github.com/radio24/TorBox/files/14140613/240202_ZeroUSB.zip)

You may need supports (gray) in the USB-A connector cutout and the four nut cutouts, see below. 

[image]

radio24, 2024-02-04:
Great work. Again, in case you missed it: With your contribution, you earned a [ProtonMail $100 Gift Card](https://shop.proton.me/collections/coming-soon/products/proton-100-gift-card-digital) from our "feedback-giveaway" program. Please send a message to the TorBox email.

gozillah, 2024-02-06:
> I successfully connected Windows 10 via TorBox mini to check.torproject.org (only once, see below). Let me remember what I did:
> 
>     * I installed the RNDIS/Ethernet driver from [link](https://modclouddownloadprod.blob.core.windows.net/shared/mod-duo-rndis.zip)
> 
>     * I installed Bonjour from [link](https://support.apple.com/kb/DL999?locale=en_GB)
> 
>     * I connected TorBox mini to an USB port of my Windows 10 computer, whereupon Windows detected a new, unidentified network without Internet access
> 
>     * I was then able to access the Torbox mini console through Putty / ssh [torbox@torbox.local](mailto:torbox@torbox.local) and configured wlan0 to the external Internet access point. Strikingly it took Putty severeal seconds to open the terminal window.
> 
>     * Trying to connect to the new network, I opened the Windows Ethernet network settings where I observed the built-in "real" Ethernet network of my computer as well as the new network
> 
>     * Opening the new network settings I disabled DHCP and manually entered the IPV4 network params (IPV6 off)
> 
>     * In this network adapter's properties I allowed others to share this "Ethernet" connection and activated all services (1700 ff, any one really used for Internet access?)
> 
>     * In the extended options I entered the network address 192.168.44.1 (you could also do it in the device manager, RNDIS/Ethernet Gadget extended driver options)
> 
> 
> Unfortunately I was unable to repeat this success ...

I finally managed to access the Internet from my Windows PC via my USB-connected TorBox mini and its wlan0 to my private router: In Windows' firewall settings (Settings\System and Security\Windows Defender Firewall\Authorized Apps) I had to allow Windows' Bonjour service access to public networks. Windows seems to categorize TorBox as a public network. However the consequences of this adjustment in terms of security are not entirely clear to me ...

gozillah, 2024-02-08:
Improved 3D design: [240208_ZeroUSB.zip](https://github.com/radio24/TorBox/files/14210639/240208_ZeroUSB.zip)

- New trick to make the slicer route a star around the screw boreholes, making supports inside the screw and nut recesses unnecessary. On my printer I can now completely renounce on supports, even in the USB-A cutout.
- Aligning four floors of holes during assembly is a bit intricate. I added alignement rails for the USB adapter, so there are now only three initially misaligned floors.

gozillah, 2024-03-07:
I have been testing 32bit Bookworm Torbox053 mini on Pi Zero 2W via USB to Windows 10 and MacOS Monterey for quite a while now. It worked very reliably. Connecting it to Android and iOS is still uncharted however. 

Do you intend to include TorBox mini in the next release image or as a separate image? 

How could I help? My core area being 3D design rather than Pi and networking: Would it make sense that I scripted instructions about where to purchase the USB adapter, howto 3D print and assemble a compact enclosure and configure Windows and MacOS? 

Additionally I designed a compact and mature 3D printable enclosure for an autonomous TorBox WiFi access point based on Pi Zero 2W, second WiFi interface and Li-ion battery for 2 hours of stand-alone operation, using commercial off-the-shelf components (with the exception of the enclosure). It will be important for this configuration to maintain the loading of the various WiFi drivers as part of the new release.

radio24, 2024-03-26:
Finally, I'm back again from my break - at list I should be during the Easter weekend. :-)

Sorry for the break, but I had to concentrate on my regular day job.

Anyway, I integrated the branchs for "[TorBox on the Cloud](https://www.torbox.ch/?p=3522)" and "TorBox mini" into the development branch "[torbox_v054](https://github.com/radio24/TorBox/tree/torbox_v054)". I will work on that branch during the Easter weekend according to the todo list, which I transferred to the [discussion section](https://github.com/radio24/TorBox/discussions/276).

gozillah, 2024-03-27:
This is very good news, thank you!

So far I used Pi4+, Zero W and Zero 2W, But I now own a Pi5. Would it help if I installed 054 on the Pi5 and did some basic testing? Or do you have test scripts?

radio24, 2024-03-27:
Indeed, testing the "[torbox_v054](https://github.com/radio24/TorBox/tree/torbox_v054)" branch would help me a lot. I am particularly concerned with the following questions:
- Are there possibilities to make TorBox more user-friendly?
I suddenly experienced problems starting up the TorBox with Raspberry Pi 5. Maybe the SD Card in the Raspberry Pi 5 is more prone to fail when pulling out the power before shutting it down properly. 
I also encountered some disturbing situations after installing the additional network drivers. Once, it seemed that the driver wouldn't work, but after trying it again, it worked fine.
I'm also unsure if the additional network drivers should already be installed in the provided image.

I've set up a TorBox chat on my TorBox on a cloud test version: mmx32tjdurtxjrnsh63skku7mg6hdccfzitauppm5y4b2ligqqdqucad.onion. This is another platform where we can discuss and address any issues you encounter during testing. You (and others) can meet me there.

The same server also runs a test version of the TorBox file server (TFS) with the latest TorBox image: x63xkeiw3cgczc6lcwf62aoe35rp6hfcz3mympmuha7xhj63qdf3ngid.onion
The specific TFS is still based on the older version 1. We have to test the newer version 2, created by Zotil, later.

gozillah, 2024-06-04:
Sorry this was too fast. I could access the GeeekPi TorBox mini via USB, but it made its Internet connetion through the preconfigured wlan0, not eth0. I will try to enable eth0.

PapiJalopy, 2024-07-30:
@radio24 Just wanted to note here that the latest bookwork release does not allow me to ssh into the Raspberry Pi Zero 2 W. SSH hangs and fail to connect. Taking the same sd card and putting it in a PI 5 works fine. As far as i know there is no issues with the bullseye 0.5.3v release. I'm able to use it on the Pi Zero 2 W just fine. 

Also wanted to request an audit on the "Build from source" documentation page, as its quite outdated and following the guide results in a broken build. I would love to test the latest revisions and other branches but I'm not sure how to correctly compile a working build.

I would also like to note that ssh was functional when I built the master branch from source. But a lot of stuff was not functional and isc-dhcp-server was failing to start.

PapiJalopy, 2024-07-30:
@radio24 Just wanted to note here that the latest bookwork release hangs for about 10-20 minutes before allowing me to SSH into a Raspberry Pi Zero 2 W.  I did not have the same issue with the bullseye 0.5.3v release.

Also wanted to request an audit on the "Build from source" documentation page, as its quite outdated and following the guide results in a broken build. I would love to test the latest revisions and other branches but I'm not sure how to correctly compile a working build.

I would also like to note that ssh didnt hang for that long when I built the master branch from source. But a lot of stuff was not functional and isc-dhcp-server was failing to start.

radio24, 2024-12-08:
Only for the record: the Raspberry Pi OS release from November 2024 removed the `ifupdown` package from its latest images, which breaks the headless configuration of the Raspberry Pi Zero 2 W by altering `cmdline.txt` and `config.txt `as well as using `firstrun.sh`.

Therefore, we recommend configuring a TorBox mini SDCard using a Raspberry Pi with the installation script and the `--trobox_mini` option. For details, see [here](https://www.torbox.ch/?page_id=3544).

closed 2024-12-08 by radio24