zotil/TorBox #277: Missing IP Address

gozillah, 2024-03-27:
When starting the wireless manager from the main menu after boot, the wireless menu will as per default automatically connect to the most recent WiFi AP (if available), showing its IP address in square brackets. However the wireless menu will sometimes only show a "-" in square brackets.

[image]

radio24, 2024-03-27:
Yes, that's the situation when the WiFi is connected, but the interface doesn't get an IP address. There are different reasons for this. I don't know if Zotil can mitigate this situation. He may have time to look at it and let us know.

zotil, 2024-03-27:
> However the wireless menu will sometimes only show a "-" in square brackets.

Thanks for flagging this. The "-" signifies that the IP address is being retrieved in the background. Perhaps we could consider changing it to "Obtaining IP" for clarity.

closed 2024-04-07 by radio24