zotil/TorBox #295: Modifying debian/mini script to create fully open source TorBox-Mini using an Orange Pi Zero 2

PapiJalopy, 2024-08-01:
If you have time it would be beneficial to the open source/foss community and individuals if you could make a modified version of the Debian install script to include configurations to produce a torbox-mini variant for Orange Pi Zero 2. 

The Raspberry Pi is not fully open source and parts of it such as the Broadcom SoC, GPU firmware, camera and display interface, first and second stage bootloaders, firmware blobs, and binary only drivers, as well as parts of the kernel are proprietary. 

Ill be happy to help in any way I can. 

From what I understand that needs to be done is adding the config.txt and cmdline.txt parameters and modifying the script to point to the "mini" configured files such as interfaces, dhcpd, iptables, rclocal, etc. or possibly just modifying the run_install_mini.sh script instead of the Debian script and removing any raspberry pi specific kernel updates and stuff like that and changing the repositories to Debian ones. I'm not sure if the first run script applies to a normal Debian install but if it can we can possibly edit that too.

Ill try to start working on it and see if can hack together both the mini and Debian scripts to make a functional script that will work with the orange pi. To my understanding on the Orange Pi its just a bare bones Debian install but the same functionality with RNDIS and CDC_ETHER will still work. If it comes down to it ill do a manual install once I get the Orange Pi Zero 2 W in the mail just as a proof of concept.

radio24, 2024-09-18:
If you download a raw Debian image for Orange Pi Zero, TorBox's Debian install script should work without problem.

We are currently focusing on the TorBox mini based on Raspberry Pi. If we are finished successfully integrating the Raspberry Pi Zero 2 W specific code into the main TorBox install script for Raspberry Pi OS, then we can migrate the changes to the Debian install script.

The question is if this is necessary. According to [this article](https://raspians.com/how-to-run-raspbian-on-an-orange-pi/), Raspberry Pi OS is also running on some Orange Pi modells.

PapiJalopy, 2024-09-19:
If possibly I have no problem using Pi OS, I have been testing the images provided by OrangePi, and have come across an "issue" that may cause a slight inconvenience.
[image]

Only the top left USB0 port can be changed between host and device mode, while the right USB1 is fixed as host mode. 

This means that USB adapters such as these ones -

[image]

[image]

will not work because they are piggy backing off of the USB1 pads on the rear of the device.

radio24, 2024-12-30:
We will adapt the installation script for TorBox on Debian soon so that you can do a test with Orange Pi Zero 2.
We will let you know as soon as it is ready.

PapiJalopy, 2024-12-30:
Sounds good though I'm not 100% certain a raw Debian install will work with a Orange Pi Zero 2. 
Normally a Armbian image is used, or the image that Orange Pi offers themselves.

radio24, 2025-03-23:
We needed some time to deal with your issue, and unfortunately, I have bad news.

I agree that a raw Debian will not work on an Orange Pi Zero 2. There is not even a version working with a Raspberry Pi 5 until now. The following logic choice for an Orange Pi Zero 2 is [Armbian](https://www.armbian.com/orange-pi-zero-2/). Because I don't own an Orange Pi Zero 2, I tested the [Armbian 25.2.2 Bookworm Minimal / IOT release ](https://www.armbian.com/rpi4b/) with a Raspberry Pi 5. The Debian installation script worked without any problem, but already before the installation, the Armbian installation didn't seem to support WiFi USB adapters. I tried to install additional network drivers, but it didn't work. The same is true with the Armbian 25.2.2 Bookworm Server /CLI Image.

I will not look further into the issue because the Orange Pi Zero 2 hardware is too selective. Also, Armbian's hardware support seems limited, which may be changed. However, searching for solutions and implementing them is too time-consuming for me. There is a reason that all the scripts are in TorBox's GitHub repository, and t[here is a detailed manual on how to build TorBox from scratch](https://www.torbox.ch/?page_id=205): someone can adapt the scripts to his unique situation and even share it with others.

PapiJalopy, 2025-03-23:
I understand your decision, though I just wanted to state that the Armbian/Debian image provided by OrangePi themselves are a lot more complete, including most needed drivers, applications and network/system configurations already preinstalled compared to a bare stock Armbian image. 

From what I understood when playing around with it, Armbian is barebones and requires a lot of user configuration and setup, I remember i even had issues getting RNDIS/CDC_ETHER and network working, probably due to missing configuration or ACPI issues. 

On the OrangePi provided images it worked OOB. 

Anyways maybe in the future Ill work on this myself because the idea is still good (having fully open source hardware and software solution) but at this time i agree that it is too time consuming.

PapiJalopy, 2025-03-24:
@radio24 I'm posting this here mostly for me, though If you every decide to come back to this in the future, here's a few resources that should help clear up the external wireless adapter issue you were facing, as well as a small Orange Pi Zero Armbian setup guide.

It looks like Armbian is a little bit selective on supported wifi chipsets, and it seems you got a unlucky and just so happened to have an adapter with a unsupported chipset. 

https://forum.armbian.com/topic/26406-usb-wifi-adapters-that-are-supported-out-of-the-box/
https://teamgloomy.github.io/armbian_wifi.html
https://github.com/bassamanator/armbian-orangepi-how-to
http://www.orangepi.org/orangepiwiki/index.php/Orange_Pi_Zero_2W

radio24, 2025-03-24:
Did you ever try DietPi Orange Pi Zero 2? Currently, I'm doing a test installation with TorBox's Debian installation script.

PapiJalopy, 2025-03-24:
I haven't tried it. but I've heard of it. I think someone posted an issue on here recommending it. Ill check it out. Let me know how it goes. It could be promising if it works.

radio24, 2025-04-10:
The TorBox Debian installation script is now DietPi-ready. Of course, more tests are needed to determine if all functions are working, but it looks good. I would appreciate more tests and feedback on TorBox running on DietPi. More about the installation procedure: https://www.torbox.ch/?page_id=1168#script

PapiJalopy, 2025-04-11:
Awesome to hear! I'll test it out when I have some time and post here with any issues I find.

PapiJalopy, 2025-05-11:
@radio24 Hello, I just got around to testing TorBox Debian install script on my OrangePi Zero 2W. I installed the script using the --torbox_mini flag as this was the use case I had in mind for this SBC.

From the beginning, I had trouble installing TorBox over the onboard WiFi via SSH. Connection was very unstable, WIFI driver would crash constantly and at time cause a kernel panic that would complete crash the OS. These are the errors I got when the WIFI driver had crashed.

`sprdwl:sprdwl_fc_add_share_credit, 536, mode:1 closed, index:0, share it`
 `unisoc_wifi unisoc_wifi wlan0: sprdwl_report_connection "MY WIFI BSSID" failed status code:1!`

From the small amount of research I did it seems that WIFI drivers for the Allwinner AW859A (aka CDTech 20U5622) are very unstable and a lot of people are reporting similar issues.

I ended up switching to a ethernet hat to complete the install.

Once the install was complete. I went through the first time setup script. Unfortunately, none of the network adapters are recognized by your scripts. Both `eth0` and `wlan0` options cause the script to display the `There is no network interface to configure!` popup.

Checking `ip addr` or `ip a` at first showed only eth0 and eth1 network adapters. I had to manually load WIFI drivers using `sudo modprobe sprdwl_ng`. Once loaded `ip addr` showed `eth0`, `eth1`, and `wlan0`. I again ran `./menu` and tried to configure both `eth0` and `wlan0`, but both failed with the same `There is no network interface to configure!` popup.

It seems network interfaces are not recognized by your scripts. At the moment I have not looked into it any further than this. If you have any input on this I would appreciate it.

PapiJalopy, 2025-05-12:
Hello, I ended up reinstalling OS and re-running the Debian install script. But before I did, I had to added, `g_ether` and `sprdwl_ng` to `/etc/modules-load.d/modules.conf` to load the the USB ethernet host mode module as well as the WIFI module.

For some reason, WIFI interface is not loaded automatically .
Without both of those modules loaded, `wlan0` and `usb0` interfaces are not visible in `ip addr`.

With both of these loaded. Option 7 of the `./menu` script is now functional, allowing to connect a wireless network and pipe it via `usb0`. I believe not having these modules loaded was the cause of the `There is no network interface to configure!` pop-up

I am now facing a couple other issues, regarding the `lo` and `eth0` interfaces.
The `lo` or local interface is in a `UNKNOWN` state. This seems a bit odd to me and felt it should be mentioned. 

When selecting Option 5 in the `./menu` script, it now allows me to proceed through the setup without the `There is no network interface to configure!` pop-up, but once it gets to the TOR connection test, it fails to connect requiring me to press `q` to quit.

When selecting the Option 5, is the script automatically supposed to update the `/etc/network/interfaces` file?
Monitoring the `/etc/network/interfaces` file, contents do not changed and only stay in the "mini" configuration.

I also want to mention that `/etc/network/interfaces.d` is empty, not sure if that matters.

If needed I can still send the requested files.
I just wanted to update you on the situation because it has changed.

radio24, 2025-05-12:
TorBox mini on Debian can be installed with the command `run_install_on_debian.sh --torbox_mini`. It adds the RNDIS/Ethernet Gadget mode. In this case, adding `g_ether` to `/etc/modules-load.d/modules.conf` shouldn't be necessary. With that option, TorBox and some configuration files will be installed explicitly so that the Internet is assumed to be on `wlan0` and a client on `usb0`. 

TorBox on Debian can be installed with the command `run_install_on_debian.sh`. In this configuration, you can freely choose where the Internet is coming from and which interfaces you want to connect your clients, except `usb0`, because RNDIS/Ethernet Gadget mode will not be supported. I thought that's what you wanted.

Theoretically, to add support for freely choosing the Internet source and the support that clients can be connected on several interfaces in parallel, should be possible with a little bit of work. Choosing entry 7 in the Main Menu could not only work for clients on `usb0`, but clients could be connected on `wlan1` (`192.168.42.1`), `eth0` (`192.168.43.1`) and `usb0` (`192.168.44.1`). The basics for the multiple client support on TorBox mini are already in the menu script, but I disabled the respective network reconfiguration because I couldn't test it. That's why `/etc/network/interfaces` didn't change in your test.

If you are interested, I can add/activate this part in a separate branch in the next few weeks.

PapiJalopy, 2025-05-12:
I still havent tested if hotspot mode works on this wifi chip so that would have to either be tested by installing without the `--torbox_mini` flag or when you finish the separate branch

radio24, 2025-05-30:
> It did not automatically enable 'RNDIS/Ethernet Gadget mode" aka "device mode" and required me to add the `g_ether` module manually.
I found the problem. Starting with Debian 12 (Bookworm) and newer Raspberry Pi OS, the standard paths for `cmdline.txt` and `config.txt `have shifted to `/boot/firmware/cmdline.txt` and `/boot/firmware/config.txt`. However, DietPi still relies on the old `/boot` paths for compatibility. The issue is fixed with commit [a670ad5](https://github.com/radio24/TorBox/commit/a670ad569e3963bf2f47b7afc1a49b42b62d5c98).

radio24, 2025-05-31:
By the way, I started yesterday with the multiple client support on TorBox mini. First commits to the [mini_multiple_client_support branch](https://github.com/radio24/TorBox/tree/mini_multiple_client_support) should follow this weekend.

radio24, 2025-06-22:
Thank you for testing the multiple client support in TorBox mini. 

> When selecting option 7, it defaults to eth1 instead of eth0, which seems unintentional. Options 5 and 7 are different. Option 5 supposes that the Internet is coming from eth0 because if you have only one Ethernet interface, you will plug in the connection to your Internet router. If you have two Ethernet Interfaces, we still suppose eth0 is the Internet and eth1 is the client. However, with Option 7, it is supposed that the Internet is coming from wlan0, which means that if an Ethernet interface (eth0) is available, most likely a client will be connected. Because of some remaining old code, it could be that it didn’t work correctly. I fixed that by removing the old code. Also, I updated the dialogue windows a little bit, which hopefully will be clearer for the user.

> I would like to also mention when switching between options, usb0 link is restarted, resulting in SSH disconnection.
It should be fixed now.

radio24, 2025-06-23:
If the Ethernet hat is shown as eth1 and you want to use it as an Internet source, why not use option 9 (eth1 --> wlan0 / eth0 / usb0 / tun1)?

radio24, 2025-07-11:
Regarding `usb0` and `hostapd` disabling: Yes, after activating `hostapd`, `usb0` was configured as a DHCP client in `/etc/network/interfaces`. However, that should change after selecting the Internet source in the Main Menu the first time after enabling `hostapd`. Of course, if `hostapd` is not persistent after rebooting the system, this will be annoying because it also removes the static IP for `usb0` in `/etc/network/interfaces` every time. The reason for disabling `hostapd` during a reboot was due to the incorrect position of `sudo systemctl daemon-reload`.

Regarding the SSID change issue: I guess this is related to the problem above. I assume that after restarting the system and finding out that `hostapd` was disabled, you enabled it again by using the Configuration Menu, which will not only enable `hostapd`, but also restore the default `hostapd` configuration. Fixing the problem above should also fix your SSID change issue.

> Also trying to deactivate other connection options doesn't seem to do anything visually. 
You have to add `TORBOX_MINI_DEFAULT=1` in the `/run/torbox.run` file manually.

Commit [e042ff2](https://github.com/radio24/TorBox/commit/e042ff2ede3155a89600e6b32de68f417e22e9e8) will fix the problems mentioned above. Nevertheless, choosing the Internet source after enabling `hostapd` might still be necessary.

radio24, 2025-07-12:
Ok, I guess because of the updates something is messed up with the configuration. Please sende me the following files and information, so that I can figure out in the next hours what’s the problem:
- output (copy-paste) of the command `ifconfig`
- the output (copy-paste) of the command sudo `iptables-save`
- the file `/etc/network/interfaces`
- the file `torbox/run/torbox.run`
- the file /etc/tor/torrc

You can send me this information via email ([anonym@torbox.ch](mailto:anonym@torbox.ch))

radio24, 2025-07-12:
Thank you!
Ok, `/etc/tor/torrc` looks fine. The other files, however, are pointing to different Internet sources. As I suspected, the updates, bugs, and some of my fixes messed up your configuration. I will review my thesis and the code again. Afterwards, I will send you the fixed configuration files to reset the system to its default TorBox mini configuration (but with the full connection options in the Main Menu). From there, you can first update the code and hopefully successfully configure your system as you want it to be.

radio24, 2025-07-12:
`cd "$(dirname "$0")"`: Good point! I should have thought about the occasion someone would start the menu from another folder. 🤦 I will add that to all menu-scripts.

radio24, 2025-07-12:
Branch [mini_multiple_client_support](https://github.com/radio24/TorBox/tree/mini_multiple_client_support) is updated by Commit [5e2a069](https://github.com/radio24/TorBox/commit/5e2a069923826960fe94cdf8995323a272b062a6) and [313a6be](https://github.com/radio24/TorBox/commit/313a6be2181b5c930f89a9bd0d096eaafe2a1aa6).

closed 2025-07-17 by radio24