zotil/TorBox #302: Suggestion: replace Raspberrypi OS for Dietpi

fclorenzo, 2024-08-23:
I'm suggesting this because:
1. Dietpi has lower resource usage than raspberry os lite (https://singleboardbytes.com/3299/dietpi-vs-raspberry-pi-os-which-is-the-better-choice-for-your-pi.htm);
2. Dietpi has less software installed by default, thus lowering the attack surface;
3. Dietpi is lighter in size, making the images smaller;
4. Dietpi is very easy to setup, and it supports automated installs with easy.

radio24, 2024-09-18:
Because DietPi is based on Debian, TorBox may be installed on DietPi by using [TorBox's Debian installation script](https://github.com/radio24/TorBox/blob/master/install/run_install_on_debian.sh).

radio24, 2025-03-24:
I tested current [TorBox's Debian installation script](https://raw.githubusercontent.com/radio24/TorBox/master/install/run_install_on_debian.sh), and it worked without Problem. However, Dietpi has changed some of the Debian tools TorBox might use and has some Dietpi-specific configurations (for example, an admin user "dietpi"). 

If you install with our [Debian installation script](https://github.com/radio24/TorBox/blob/master/install/run_install_on_debian.sh) TorBox on a DietPi installation, test it and send us a list of issues, we will work on adapting the Debian installation script so that it will work with DietPi even better.

radio24, 2025-04-10:
The TorBox Debian installation script is now DietPi-ready. Of course, more tests are needed to determine if all functions are working, but it looks good. I would appreciate more tests and feedback on TorBox running on DietPi. More about the installation procedure: https://www.torbox.ch/?page_id=1168#script

closed 2025-04-22 by radio24