zotil/TorBox #319: Permission issue during debian install script

jimmyp98, 2024-10-09:
It specifically states not to use sudo during install, but within 10 seconds im getting frontend lock permission issues, are you root? When i run the installer using the root account it installs fine, but of course im weary of this. Why is this happening on the non root account? I have sudo permission in groups etc. Everything else ive followed and double checked, but the installer wont work when not using sudo or some sort of elevation.

radio24, 2024-10-10:
[Here](https://www.torbox.ch/?page_id=1168) it is stated for Debian and Ubuntu systems:
_To install TorBox on another system or hardware, download the OS image, transfer it to an SD Card. Under Debian, log into your server as root. With Ubuntu, log into your server as a user (for example ubuntu or create the user torbox). Then download and start the installation script:_

> When i run the installer using the root account it installs fine, but of course im weary of this.
Why are you weary of this?

jimmyp98, 2024-10-10:
Apologies i didnt see that part in the guide. i was definitely needing sleep! I was weary of using the root account under the assumption i was not to use any root privvelages of any kind, but now that it states its required im not weary. I thought maybe it would have impacted security as it was doing something not supported by the project.

I now seem to be stuck at a part in the install where its asking for a new root password. It keeps spamming "Failed to start hostapd.service" over and over. The install has been running for a long time, but keeps stopping at this point as i cant go any further with the repeated popups.

radio24, 2024-10-10:
I might add a check if the installer uses the correct user.

radio24, 2025-04-21:
I just tested the script. It functions as expected when setting a new root password.

closed 2025-04-24 by radio24