zotil/TorBox #320: 502 Bad Gateway

yungestdev, 2024-11-17:
I've been trying to setup torbox on a Raspberry PI 5 (8GB ram version), everything went fine until the Web connection on 192.168.42.1, when i connect to the page it pops a 502 bad gateway (nginx), i've been looking into it but i can't find what's causing it.

yungestdev, 2024-11-18:
unluckily i can't connect to SSH, for some reason it tells me the password is wrong (CHANGE-IT)

yungestdev, 2024-11-18:
ok after some time i tried re flashing the image to the sd card (this time using etcher). now it's working as expected. if it happens to someone in the future make sure they flashed with etcher instead of raspberry imager <3.

closed 2024-11-18 by yungestdev

radio24, 2024-11-19:
Thank you for the feedback. 
We will remove Raspberry Pi Imager from the recommended method to flash the SD card.

radio24, 2024-12-06:
We did some more test. It seems that the Raspberry Pi Imager [version 1.8.5](https://www.raspberrypi.com/software/) is more reliable then the [version 1.9.0](https://github.com/raspberrypi/rpi-imager/releases/tag/v1.9.0).