zotil/TorBox #322: Tor bootstrap takes an abnormal amount of time, plus some configuration issues with services

LuckySkullThe420th, 2024-11-22:
I've been messing around with my manual build of torbox all day & night trying to connect to the internet, but I've had no luck. I've had a few problems with my tor@default.service & isc-dhcp-server.service because both of them would not start upon reboot. I learned that my IPs were not set & they never set upon boot. They disappear after a reboot, then I'm subject to the horrors of these failing services again. After successfully starting up both, I was finally able to restart & begin the bootstrap for tor, but even as we speak the connection is stuck at 0%... I've checked the logs & everything seems fine.

radio24, 2024-11-24:
We are investigating it. 
Raspberry Pi OS apparently removed the `ifupdown` package from its latest images, which caused these problems. We are currently trying out the latest changes and will post more information here if our tests are successful.

LuckySkullThe420th, 2024-11-25:
Ok, thank you.

closed 2024-11-27 by LuckySkullThe420th