zotil/TorBox #332: Torbox Mini compressed image corrupt?

airgap0, 2024-12-19:
When one tries to flash torbox mini to a SD RasperryPi-Imager v 1.8.5, Balena Etcher portable 1.18.4 and 7Zip 23.01 (x64) report a compression error. So it seems torbox-mini-20241208-v054.img.xz has an issue although the hash is identical to the published hash. See attached screendump. 
[image]
Therefore one can safely assume the download itself was free from transmission errors.

radio24, 2024-12-19:
Thank you for the feedback.
I don't know what, but something went wrong with the compression.
The file will be replaced as soon as I have built a new xz-file, tested and uploaded it.
I will make sure to keep you posted in this thread.

radio24, 2024-12-19:
The new image file is now online: [TorBox mini v.0.5.4 (08.12.2024)](https://www.torbox.ch/data/torbox-mini-20241208-v054.img.xz), the [SHA-256 values](https://www.torbox.ch/?page_id=1128) is updated.

**As information**: Balena Etcher and the Raspberry Pi Imager refuse to write the `.xz` file. However, unpacking with 7Zip should work now without a problem. See attached screen dump.
[image]

airgap0, 2024-12-20:
New xz does the job - Thank You

closed 2024-12-20 by radio24