zotil/TorBox #334: Support with eth1 network configuration

eric9000k, 2024-12-28:
I have a working Torbox on rp4B, thankyou for this.

I can not get the correct configuration of network settings during os install of ubuntu server (24.04.1).  It keeps rejecting at the mirror test screen, I have tried many configurations, and I can't get it.  Difficult to find help with this but it seems like it would be trivial to someone here.

Most tutorials I have found and chatgpt are recommending I assign a static ip in ubuntu during o.s. install, when I do this manually, I need four main values **(subnet, address, gateway, nameserver)** and then the next page needs **proxy settings.**  The problem is basically there are zero tutorials that deal with the specific issue of installing ubuntu server os under a preconfigured torbox and setting up the network config properly out of the gate.

I have also tried assigning a static ip from my isp router to the torbox but it of course randomizes at startup and asks for dynamic ip again so I have not had luck with that.

When I use automatic dhcp settings at o.s. install it fails at the mirror download with or without proxy settings although it does seem to detect a network so I think it's being blocked somewhere.

The last thing I have tried was

> subnet 192.168.2.0/24
> address 192.168.2.11
> gateway 192.168.2.0
> name servers 192.168.2.0
> proxy 192.168.43.1:9050

however to be honest I have tried a million other things as well including 127.0.0.1 and 0.0.0.0 ; I'm lost.  Thank you to any who are willing to help.

hal9000

hal9000node, 2024-12-29:
Connected the eth1 to a different client machine running a fresh install of Mint;

**Ssh not successful over eth1**
WLAN Tor settings good; ssh over WLAN good.

**network config was still not successful.**  Pulled some parameters for the ip and subnet, 192.168.43.1, but still not actually connecting to the internet.  **What is blocking the connection?**

radio24, 2024-12-30:
Zotil did a test installation on the latest Ubuntu version and discovered some compatibility problems:
- Unlike Raspberry Pi OS and Debian, Ubuntu comes with Python 3.12. We already fixed some minor problems but must look into the consequences.
- SSH is only available through key authentication, not passwords. This has been the case since the Ubuntu image was flashed. I attempted to enable password authentication in sshd_config, but it didn’t work for some reason. That's not what we want, and we will look into it.
- Tor is running and connected. However, Zotil couldn’t connect to Tor from a TorBox client. Despite the laptop being connected to the TorBox network, it couldn’t connect to port 9050. That's not what we want, and we will look into it.

We need time to sort out all the compatibility problems with Ubuntu 24.10. What can you do in the meantime?
- Using Ubuntu Server 22.04.2 LTS or Ubuntu Server 23.04, both tested with TorBox and should work without issues.
- Using Debian 12 (Bookworm)

hal9000node, 2024-12-31:
Respectfully, a fresh install ubuntu server 22.04.5 LTS had similar issues just now.  Detected enp0s31f6, assigned 192.168.43.1 but could not connect to mirror with default settings.  Tried Proxy settings 192.168.43.1:9050 which also did not respond.

Not sure if this is relevant but Tails OS will not connect to Tor network without a bridge when connected to my Torbox over the WLAN.

My hypothesis at this point is I should configure bridge settings in the Torbox UI and try again with Ubuntu?

radio24, 2024-12-31:
If there is a device `enp0s31f6` which should be `eth1` (on the device with TorBox installed), my question is if there is another device name where the Internet is connected. It would be helpful for me to get a `ifconfig` output from the device after TorBox is installed. This information is essential to locate the problem because TorBox doesn‘t support Predictable Network Interface Names - we need network interface names `eth0`, `wlan0`, `wlan1` etc. We started to address this issue (see [here](https://github.com/radio24/TorBox/blob/a36ac8dcc73735adb66d3974289ac4992e6f9458/install/run_install_on_ubuntu.sh#L1110)), but maybe it wasn‘t enough.

Please also send me the following configuration files:
`/etc/network/interfaces` —> to check why you can‘t connect `eth1` (on the device which has TorBox installed)
`/etc/tor/torrc` —> to check your proxy issues

If you try to reach a mirror from the client device through TorBox by using `/etc/apt/sources.list`, then most probably the packet manager on the client will use `http://` to fetch the packages. However, because of security considerations, by default, `http://` requests through TorBox are blocked, and you have to enable it by using menu entry 15 in the [configuration sub-menu](https://www.torbox.ch/?page_id=875).

Yes, Tails over TorBox works only with bridges. It is because tor doesn‘t allow a tor over tor scenario. See [here](https://www.torbox.ch/?page_id=112#when-i-start-the-tor-bowser-or-when-i-open-a-window-with-tor-on-brave-both-running-on-a-client-device-of-the-torbox-i-cannot-connect-to-the-tor-network-the-same-happens-if-im-using-tails-behind-a-torbox) for details

hal9000node, 2025-01-04:
[interfaces.txt](https://github.com/user-attachments/files/18307787/interfaces.txt)
[torrc.txt](https://github.com/user-attachments/files/18307788/torrc.txt)

Hello sir.

Thanks again for the assistance with this issue.   I have tried various things that nearchos has told me on youtube as well, we have determined the client machine does detect dhcp ip 192.168.43.1x at the above named port which we are calling eth1.

I can confirm I did configure the Torbox to act as a bridge relay to test one hypothesis but this did not fix my problem.

I did disable the http block as described, however it has not completely fixed the issue.  As instructed, I did attach the two files.

I will not be able to try anything for the next three weeks, but this is important to me, I would ask you to please be patient while I am away and not close the thread.  Thankyou.

radio24, 2025-01-04:
Regarding into the two files, I can tell you the following:
- Currently, it seems that you don’t use the latest version **released on the 30th of December.** Please use the newest version because some network-related configurations have been fixed or/and improved. For example, in the `torrc`, `#SocksPort 127.0.0.1:9050` should be `SocksPort 127.0.0.1:9050`, which was an error in the previous version.
- **The interface configuration is correct**: TorBox expects an Internet connection through one of the following devices: `eth0`, `wlan1`, `usb0` using DHCP to get an IP address. It also expects that the client devices are connected through the interfaces `wlan0` (`192.168.42.1`) or `eth1` (`192.168.43.1`).
- According to the `torrc`, `eth1` was **not appropriately configured** through the main menu using **entry 5**. I guess the Internet comes from `eth0` because why else would you use `eth1` instead of `eth0` for the client connection? As listed in the table [here](https://www.torbox.ch/?page_id=775), currently, that’s the only possibility to use a client on `eth1`. By using main menu entry 5, the script will test if `eth1` is present and connected with a client. Only in this case `torrc` will be configured accordingly. In other words, you have to connect the client via an Ethernet cable to `eth1` and then use the main menu entry 5. I guess you don’t use a pre-2010 computer as a client because, in that case, you most likely need a crossover Ethernet cable.

Still, an `ifconfig` output from the device after TorBox is installed would give me more insights into the potential problem.

radio24, 2025-01-11:
@zotil did a lot of tests on Ubuntu using TorBox's installation script for Ubuntu and integrated some critical fixes.
I hope this will fix your problems with TorBox on Ubuntu.

hal9000node, 2025-01-12:
I appreciate the support very much.  It is still going to be some time before I can get back to the system and try again, but I am completely optimistic.  Thanks for providing f.o.s.s. and helping me solve my issue ! 💻💾

radio24, 2025-01-12:
Today, I used [TorBox's installation script for Ubuntu](https://github.com/radio24/TorBox/blob/master/install/run_install_on_ubuntu.sh) on an Ubuntu 24.04.1 LTS. During the installation, it changed the network interface with the internet connection (ens6) to eth0, which is correct and needed. After the installation, TorBox ran smoothly. I will also test it on a local Raspi because I have to check something regarding the OpenVPN capabilities. Unfortunately, I don't have a local PC to test it.

hal9000node, 2025-01-28:
Hello again.

Thank you for being patient with me.  I regret to inform you I still cannot fix my problem.  I thought some pictures and a walkthrough would help.

1.  I obtained the newest torbox image v 0.54 of 11.01.2025 from torbox.ch
2.  (sd card in rpi) removed partitions, formatted the microsd card; decompress the file, and flashed the new image.
3.  At initial startup dialogue, selected eth0 as internet source
4.  When told to connect client device NOW, I plugged in an usb-eth dongle to the rpi and connected to the client pc
5.  switched screens and booted an image of ubuntu 22.04 LTS for fresh install on client machine
6.  Returned to raspberry pi screen and worked through basic configuration and update menu items 1-5 ; also disabled http plaintext block
7.  Numerous reboots etc - the torbox is working perfectly it seems; disconnected screen and keyboard
8.  Return to client device, proceed with ubuntu install...

[image]
**Image 1 - the client device is assigned dynamic ip 192.168.43.10/24 which is correct**

[image]
**Image 2 - Proxy settings left blank**

[image]

[image]

[image]

**Images 3-5 Mirror testing fails.**

I can not proceed until this is resolved.

Thank you for any and all further assistance you can offer with this issue.  Sincerely, Hal9000

hal9000node, 2025-01-28:
It has occurred to me we might be talking about two different things.  You may be thinking I am trying to install torbox on an existing ubuntu installation on a pc -

For greater certainty, I want to say

the **torbox** is an ordinary rpi4 that I have booted with the latest version 01.11.2025 and completed all of the updates ☚

the **client machine** is a pc that I am using to run ubuntu server os. ☚☚

I intend on using both the wlan capabilities of the torbox and the ethernet connection at the same time.  I want to run an ordinary bitcoin node on the pc and connect two miners to the wireless network and have all traffic routed through the torbox.

hal9000node, 2025-01-29:
Today I had some luck.  Very short lived.

Using menu entry 5 as you suggested, did something, that caused it to work for a very short period of time.  It passed the mirror test and fetched some data.

Then the ubuntu installation kept crashing, got hungup on reboot, and when I started over, it didn't work again.  Failed to fetch, as before.  Very disappointing, but glad to see it CAN work.

I noticed every time the computer reboots it is assigned the next ip address in sequence.  Is there a simple way to reset this counter??

I have reset torbox config files and am running through the update menus 1-5 again.  Then I will try again with the client machine to perform a fresh install once again.  I will run the first startup dialog when initializing the boot image on the client machine.

zotil, 2025-01-29:
@radio24 It could be related with #293?

If the internet was connected for a short period of time and then the connection was lost, it could be related with iptables

hal9000node, 2025-01-29:
> [@hal9000node](https://github.com/hal9000node) I will perform some tests adding an eth1 to my TorBox and installing ubuntu on an external computer to check if I get the same problem

@zotil Thank you.  If I can offer just one more clue - the time it _did_ work, there was some additional dialogue about updates during the ubuntu installation, but more importantly, the mirror location was _regional_ instead of generic.  Meaning it pinged the address http://nl.archive ... ../

This causes me to think it is the exit relay.  So I am going to try a few times with menu item2  Enforce a new exit node with new ip.

hal9000node, 2025-01-30:
> This causes me to think it is the exit relay. So I am going to try a few times with menu item2 Enforce a new exit node with new ip.

Welp.  This wasn't it.

I can not reproduce the fix.

Hopefully you are able to reproduce the problem @zotil 

I realize I can just configure bitcoin over tor and don't _need_ a torbox as part of my project.  Actually, trying to use a torbox has made it all much more complicated (I want to say for no reason but there _is_ a reason.)

402PaymentRequired on youtube has [walkthroughs for complete node setup](https://youtu.be/mbG7hBMWQrs?) with [electrum rust server](https://github.com/romanz/electrs) featuring remote access, they're awesome videos, but when I asked him for help, he didn't think I should be using the torbox at all.

Because of the [miners](https://github.com/bitaxeorg) that connect to the wlan, I really want to include this torbox.  It is a necessary part of the sovereign and autonomous home mining-node that the miners have tor-enabled privacy.  Thanks to open-source asics, [nodes can easily be miners again](https://github.com/stratum-mining/stratum).

I hope this makes sense.

hal9000node, 2025-01-30:
Today I am spinning up a fresh installation of the torbox on my rp4 and working through the update menu items again.

Then I will create a backup file and go no further.

I'll use the wlan to test out my miners but,
I will not be attempting to configure the eth1 connection or boot the PC until I hear back regarding your results, zotil. ☚

I do think this project (TorBox) could be a very crucial component to a fully autonomous and sovereign home mining-node and would really love to see a good outcome here but I am starting to think I should just include an upstream switch that sends a clearnet signal for the PC if this is too complicated, especially given what was said to me by 402PaymentRequired that it is already possible to configure the node and electrum server to work over tor without the need for the torbox.

Thanks for any further support you can offer with this issue. ☚

zotil, 2025-01-30:
> Today I am spinning up a fresh installation of the torbox on my rp4 and working through the update menu items again.

Just to be sure. You installed TorBox in your Rpi with a flashed image or installer script over other ubuntu?

hal9000node, 2025-01-30:
> > Today I am spinning up a fresh installation of the torbox on my rp4 and working through the update menu items again.
> 
> Just to be sure. You installed TorBox in your Rpi with a flashed image or installer script over other ubuntu?

Yes sir.  I obtained torbox054 01.11.2025.xz and decompressed the image then flashed it to a micro sd and installed in a rp4 this morning.

I am currently working through the update menu items 1-5 (base system, firmware, etc) and have otherwise left all configurations as basic and default (bridge relay, firewall, countermeasures, etc all false).

They say a picture is worth 1000 words so I will just show you a picture of what I have.

[image]

The rpi is just doing its thing right now updating.
The pc is OFF and has ubuntu LTS 22.04 LTS in the usb drive waiting to spin up.
The samsung t7 is doing nothing at all right now and neither are my bitaxe miners.

This is probably important and got overlooked that I was having issues with my laptop (mint) over wlan doing wget commands as well, but this was a few weeks ago.  I thought they must be related issues as I can browse the web and download files over Tor on my laptop and ssh over web browser into the torbox without any difficulty ☚

Nearchos on youtube spent some time with me trying to sort it out and we couldn't get anywhere ☚

hal9000node, 2025-01-30:
I had no problems on the laptop (mint) running Tor Browser, downloading files manually, running Tails, configuring a bridge in Tails and doing Tor over Tor on the wlan, and ssh into Torbox on wlan, all working **perfectly!**

But if I remember correctly the laptop would not fetch using wget in terminal, like I said I kind of forget now.  I have to admit I am just a pleb end user.  But I feel the issues must be related.

hal9000node, 2025-01-30:
I will try.  Please give me about an hour to try this as I am currently updating my laptop's mint os using the torbox wlan and would be interfacing with the torbox by web broswer ssh from now on.

I'm hopeful that you had success but I contend that I tried this repeatedly yesterday and had no luck after the first time.

To be clear, please let me make sure I understand you:

- Connect all cables and power on the PC
- Run menu item 5 `Ethernet cable (eth0) or TorBox on a cloud (tun1)`
**- Is there a particular screen or sequence that is important, at which point do you do menu item 5?**

zotil, 2025-01-30:
I just flashed a TorBox and started on a Raspberry 4, I used a waveshare usb/eth cap to get the eth1.

First run of TorBox all is default, when it asks where is the internet, select `Ethernet cable (eth0) or TorBox on a cloud (tun1)` which already has an arrow at the right like `<--`.

Continue until your Tor log shows 100%.

After this, your TorBox is running fine right? You have internet in eth0 and you connect a client to eth1. At this point, TorBox has not opened the Tor ports for the 192.168.43.1 IP, so your client connected to eth1 can't resolve names.

After you go to the menu and run again `Ethernet cable (eth0) or TorBox on a cloud (tun1)` it will open the ports and your client will be able to resolve.

However you will still blocked to access plain HTTP traffic. You need to disable the plain text traffic block too, in the menu 12 - 15

zotil, 2025-01-30:
> **- Is there a particular screen or sequence that is important, at which point do you do menu item 5?**

No, just re-run it

hal9000node, 2025-01-30:
> I just flashed a TorBox and started on a Raspberry 4, I used a waveshare usb/eth cap to get the eth1.
>
> First run of TorBox all is default, when it asks where is the internet, select `Ethernet cable [...]

Thank you for making it simple for me @zotil 

I will let you know very soon what happens over here.

Can you please confirm you left the proxy settings blank and just used default dynamic ip settings, sorry, thanks...

zotil, 2025-01-30:
> Can you please confirm you left the proxy settings blank and just used default dynamic ip settings, sorry, thanks...

Yes, proxy blank and all the steps by default

hal9000node, 2025-01-30:
Yes, this should be so simple.

- New torbox from flashed image, all default, updated, working good
- Disabled http plaintext block
- Login by web ssh from laptop
- All cables connected
- Power on client device
- Run menu item 5 `Ethernet cable (eth0) or TorBox on a cloud (tun1)`

<spoiler>
[image]

- Ubuntu bootup sequence

[image]
 
- Bootstrap complete

[image]

- Ubuntu detects dynamic ip 192.168.43.10/24

[image]

> Yes, proxy blank and all the steps by default

The outcome:

FAILED to fetch.

[image]
</spoiler>

- Okay, return to previous screen > return to Welcome screen on client device
- Click menu5 again, selects `open access`  [..........bootstrapped 100%], press q
- Client machine, still has correct ip
- Selects `Try Again Now`
- Mirror test fails again.
- Return to previous screen
- Return to ip config screen
- Reruns menu 5 again
- Selects `Edit ipv4 settings`, highlights `automatic`, selects `save`, ip flashes and refreshes, same ip
- Fails again

hal9000node, 2025-01-30:
Question would config menu 14 `Enable Tor access control port for clients` be relevant here?

zotil, 2025-01-30:
> Question would config menu 14 `Enable Tor access control port for clients` be relevant here?

It should not be relevant.

Can you share what does the command `ifconfig` prints in the TorBox?

zotil, 2025-01-30:
and this `netstat -an|grep tcp | grep LISTEN`

hal9000node, 2025-01-30:
@zotil

I have no idea why it worked for you and not for me.

> Can you share what does the command `ifconfig` prints in the TorBox?

[image]

> and this `netstat -an|grep tcp | grep LISTEN`

[image]

zotil, 2025-01-30:
Sorry run it without tcp

`netstat -an | grep LISTEN`

zotil, 2025-01-30:
In your ubuntu installation, open a new tty pressing alt + F2, and execute the following checklist

- `ip addr` and check your interface has 192.168.43.x TorBox IP
- Test pinging an ip like `ping 1.1.1.1` if it works, you're good until now
- Check resolving with `dig google.com @192.168.43.1` this should resolv the IP address of google.com

Let me know what you get

hal9000node, 2025-01-30:
> Sorry run it without tcp
> 
> `netstat -an | grep LISTEN`

[image]

> In your ubuntu installation, open a new tty pressing alt + F2, and execute the following checklist
>
> * `ip addr` and check your interface has 192.168.43.x TorBox IP
> * Test pinging an ip like `ping [...]

Yes, it will take some time, as I just left it hanging at the Network Config screen, I have proceeded with the ubuntu installation on the PC.

It's currently hung at `configuring apt` but I think it will eventually move things along.

I will let you know after the installation is finished.  Thank you for your help.

zotil, 2025-01-30:
> > `netstat -an | grep LISTEN`
> 
> [image]
> 

Seems that the port 9053 is not open, which will allow the client to resolv. In this case, the 3rd step I mentioned above should fail:

> > * Check resolving with `dig google.com @192.168.43.1` this should resolv the IP address of google.com

To solve this quick, in the TorBox, edit `/etc/tor/torrc` file, and check that `DNSPort 192.168.43.1:9053` line, doesn't have a **#** at the start of the line. If it has it, remove it. Save the file then restart the tor service with `systemctl restart tor.service`

Now you client should be resolving names and allow you to pass the "apt update" step

hal9000node, 2025-01-30:
Unfortunately the config file did not have # on that line, please see the image, I have not edited the file at all.

[image]

hal9000node, 2025-01-30:
@zotil 

Maybe this means something to you?

When I hit `ip addr` it gives me this output:

[image]

Is there something unusual or suspicious about this?

zotil, 2025-01-30:
Seems good to me. Next could be the iptables rules

please do a `cat /etc/iptables.ipv4.nat` and show the output

zotil, 2025-01-30:
> Unfortunately the config file did not have # on that line, please see the image, I have not edited the file at all.
> 
> [image]

With that, restart tor anyway with `sudo systemctl restart tor.service`

then have a look again to the ports open
`sudo netstat -an | grep LISTEN`

9053 must open

zotil, 2025-01-30:
> [image]

From this computer you could try `dig google.com @192.168.43.1` to test if it's resolving ok

hal9000node, 2025-01-30:
> From this computer you could try `dig google.com @192.168.43.1` to test if it's resolving ok

communications error to 192.168.43.1#53 timed out

no servers could be reached

zotil, 2025-01-30:
>
> communications error to 192.168.43.1#53 timed out
> 
> no servers could be reached

This confirms that the problem is the name resolving.

You should check in the TorBox if the port 9053 is open in 192.168.43.1 ip address with `netstat -an | grep LISTEN`. If 192.168.43.1:9053 is not listed, you should check in `/etc/tor/torrc` if `DNSPort 192.168.43.1:9053` line is commented. If it's not, restart tor anyway and check again if the port is open with netstat.

On my test, it also happened the name resolving problem, but I got it fixed going to the TorBox menu and re-run the menu 5.

Note that I disabled the HTTP plain text blocking after running the menu 5

hal9000node, 2025-01-30:
> > communications error to 192.168.43.1#53 timed out
> > no servers could be reached
> 
> This confirm that the problem is the name resolving.
> 
> You should check in the TorBox if the port 9053 is open in 192.168.43.1 ip address with `netstat -an | grep LISTEN`. If 192.168.43.1:9053 is not listed, you should check in `/etc/tor/torrc` if `DNSPort 192.168.43.1:9053` line is commented. If it's not, restart tor anyway and check again if the port is open with netstat.
> 
> On my test, it also happened the name resolving problem, but I got it fixed going to the TorBox menu and re-run the menu 5.
> 
> Note that I disabled the HTTP plain text blocking after running the menu 5

Thank you for the help, @zotil 
I will try this in about 5 or 6 hours.  I have shut the system down for a while as there be some children about.

It appears these are the same instructions we have tried previously but as you have said it worked for you, it should also be working for me.

hal9000node, 2025-02-01:
Hello.

This torbox is a very interesting project.  Unfortunately, at the current time, I can not get it working with the eth1.  I have noticed another user posted another discussion thread about dual clients and has indicated he is also not fetching packets with the eth1 port.

I suspect he is having the same underlying problem that I am.

@zotil has instructed me to use menu5 while the client device is running and this will open port 9053.

While it did work once for me quite briefly, I can not get it working again.  Whatever parameters have made it work, I can not reproduce.

Sirs, I suggest you consider this issue to be an unresolved bug in the code, with respect.

And thank you for your assistance.  I am giving up for now, but I want to thank you all the same for this, and I will use it strictly for it's wifi abilities.

radio24, 2025-02-02:
See here: https://github.com/radio24/TorBox/issues/339#issuecomment-2629560058

hal9000node, 2025-02-02:
> See here: [#339 (comment)](https://github.com/radio24/TorBox/issues/339#issuecomment-2629560058)

Looks really promising.

Thank you for taking the time.

It's going to be a couple of weeks before I can get home and try again but I am very optimistic.

radio24, 2025-02-03:
That's great to read, but according to your explanation [here](https://github.com/radio24/TorBox/issues/334#issuecomment-2620075601), you already did that. 

I suggest **to finish the TorBox installation** before testing with a client and only **after that** to start the Ubuntu image. 

Why that? The Ubuntu image will get all network configuration information (IP, gateway, route, DNS via DHCP from TorBox). If, at an early stage, this information is not given entirely or correctly to the client with the Ubuntu image, your client may not fix this automatically later. In other words, if it doesn't work, don't check the DNS information of the TorBox; check the one of your client, if that is possible. The gateway and the DNS should be 192.168.43.1 if connected via an Ethernet cable.

RampantDespair, 2025-04-27:
@radio24 
Hi, I'd like to add my experience to this issue.

I'm running the latest TorBox version on a Raspberry Pi with Ubuntu Server, and I encounter very similar reboot-related network issues as originally described here.

---

### My setup:
- **eth1** = Internet uplink (connected to main router via DHCP)
- **eth0** = LAN side (connected to a switch and client devices, providing DHCP 192.168.43.x)
- TorBox installation completed successfully, including Menu 1-5 setup and disabling HTTP blocking.

---

### The issue:
- On first setup and initial boot, everything works perfectly — client devices get DHCP IPs, DNS resolution works, internet access through Tor is functional.
- After reboot, TorBox becomes non-functional:
  - Client devices still receive IP addresses.
  - Local DNSPort 9053 **is listening** correctly on 192.168.43.1.
  - However, the **default route to the internet is missing** after reboot.
  - As a result, `ping 8.8.8.8` fails from TorBox itself.
  - Tor fails to bootstrap (no external connectivity).
  - Client devices have no internet access.
  - Running `/etc/rc.local` manually after boot does not fully restore functionality.

---

### What I have tried:
- Verified `/etc/rc.local` is executable, active, and running under systemd.
- Added a "smart wait" loop in `/etc/rc.local` to delay until both eth0 and eth1 are UP before proceeding.
- Added automatic checks and restarts of tor.service if port 9053 was not open.

```sh
# Wait for interfaces
for i in {1..20}; do
    if ip link show eth0 | grep -q "state UP" && ip link show eth1 | grep -q "state UP"; then
        echo "eth0 and eth1 are UP"
        break
    fi
    echo "Waiting for eth0 and eth1..."
    sleep 1
done

# Check Tor DNS port
echo "Checking if DNSPort 9053 is listening..."
if ! ss -ulpn | grep -q ":9053"; then
    echo "DNSPort 9053 NOT found — restarting tor.service..."
    sudo systemctl restart tor.service
    sleep 5
else
    echo "DNSPort 9053 is active."
fi
```

---

### Hypothesis:
- TorBox's reliance on rc.local scripting causes race conditions during boot?

---

### Temporary workaround:
- I have installed a UPS on my TorBox and am avoiding reboots until a more permanent solution is implemented.

radio24, 2025-04-27:
@RampantDespair 
Thank you for sharing your experiences. 

I have three questions:
- I guess that you can log into TorBox from the client after reboot. Did you then try to select and run entry 5 in the Main Menu again, even if there is already an arrow on the right side, so that the network is automatically configured and Tor is restarted again?
- Did you see any errors when running `rc.local` manually?
- Did the added "smart wait" and the check resolve your problem?

RampantDespair, 2025-04-27:
@radio24 
Thank you for following up.

1. After reboot, I access the TorBox directly via HDMI and keyboard/mouse (not over the network). I tried manually selecting and rerunning Menu Entry 5 after reboot. It did not restore functionality.

2. When I manually executed /etc/rc.local from a terminal after reboot, it ran without any visible errors (no crash, no syntax issues). However, it did not properly restore the missing routes or fix the internet connectivity — Tor still could not bootstrap.

3. Unfortunately, it did not resolve the issue. After reboot, the system consistently has:
- No default gateway.
- No internet access.
- Tor failing to bootstrap.

radio24, 2025-04-28:
@RampantDespair 
I assume you are using the latest version of the TorBox (the image from January or April 2025) and have updated the TorBox Menu by selecting [entry 5 in the Update and Maintenance sub-menu](https://www.torbox.ch/?page_id=1410).

What is the reason that you don't access the TorBox from the client? Doesn't it work with a browser or a client using the IP address `192.168.43.1`?

If I understand your setup correctly, a Switch is placed between the client and the TorBox. Did the issue also appear when the client was directly connected to the TorBox without a Switch in between?

To better understand your specific situation, it would be helpful to get from you the following information after a restart:
- the output (copy-paste) of the command `ifconfig`
- the output (copy-paste) of the command `sudo iptables-save`
- the file `/etc/tor/torrc`
- the file `/etc/network/interfaces`
- the file `/var/log/tor/notice.log`
- the file `/etc/hostapd/hostapd.conf `(don’t forget to delete your password)

You can send me this information via email (anonym@torbox.ch)

RampantDespair, 2025-04-28:
@radio24 

> I assume you are using the latest version of the TorBox (the image from January or April 2025) and have updated the TorBox Menu by selecting [entry 5 in the Update and Maintenance sub-menu](https://www.torbox.ch/?page_id=1410).

I am using Torbox `v.0.5.4`, which to my understanding is the latest version unless I am mistaken?
[image]
Also, I'm sorry I misspoke looking back, about the 1-5 menu, I meant to say that my torbox is running the latest version to my understanding.

> What is the reason that you don't access the TorBox from the client? Doesn't it work with a browser or a client using the IP address 192.168.43.1?

On first startup everything works like a charm, my physical setup permits me to access devices physically, with a keyboard and hdmi cable, that's why I do it. However I can open an ssh session from a client device to it, and yes, the dhcp leases work fine. **All of this works great on first startup only.**

> If I understand your setup correctly, a Switch is placed between the client and the TorBox. Did the issue also appear when the client was directly connected to the TorBox without a Switch in between?

The setup is:
client devices -> switch -> (eth0) torbox (eth1) -> router -> internet
Yes the issue appeared when straight piping the client to the torbox.

> To better understand your specific situation, it would be helpful to get from you the following information after a restart:

Sure I can do that, I'll give you the before and after in a few days.

RampantDespair, 2025-05-01:
@radio24 
Alright, my torbox inexplicably crashed during the night forcing me to restart.
So I sent you all the files you asked for except `/var/log/tor/notice.log`, as it does not exist.
Let me know if there is anything else you need.

RampantDespair, 2025-05-01:
@radio24 
Another interesting thing I noticed is that on first bootstrap, like right now.
It gets stuck at `Problem bootstrapping` and I need to physically unplug and replug my eth1 (ETH->USB) adapter which in turn fixes it. After this, the bootstrap goes to 100% without a problem.

radio24, 2025-05-04:
@RampantDespair 
Thank you for sending me your config files. With the files, your explanation, and some more tests, I could locate two problems, which I solved on GitHub. 

**1 - Not saved iptables during initial configuration**
In [this line](https://github.com/radio24/TorBox/blob/d583d5cbacb79cef012d159f9f0a2194fe505f77/menu#L906) of the menu script the following command was missing: `(sudo sh -c "iptables-save > /etc/iptables.ipv4.nat") 2>/dev/null`

This means the configuration was rightly set during the first start, but the modified `iptables` were not saved, causing your problem after rebooting TorBox. However, in my test, starting TorBox again, accessing the Main Menu directly with an attached keyboard and screen and choosing entry 9 in the Main Menu again healed the problem. After restarting TorBox again, the Client gets its IP (`192.168.43.11`) and TorBox is accessible via `192.168.43.1`. The Internet is also accessible from the client with Tor support - at least sometimes. The problem is that TorBox was often unable to get the default gateway configuration, causing a disconnection between TorBox and the Internet.

**2 - Missing default gateway**
You rightly mentioned that problem [in your comment above](https://github.com/radio24/TorBox/issues/334#issuecomment-2833603529). However, I don't think the problem is caused by `rc.local` - most probably the issue is already caused before, maybe during the initialisation of the network, when, for some reason, the standard gateway couldn't be identified. However, [I already had and solved this problem before on `eth0`](https://github.com/radio24/TorBox/issues/228#issuecomment-1627386773). I only had to add the same routine for `eth1.`

**How to permanently fix your installation**
- Log into your TorBox and run entry 9 in the Main Menu. With some luck, it will work.
- If Tor will not start up and cannot be restarted using entry 3, then try entry 10 in the [Update and Maintenance menu](https://www.torbox.ch/?page_id=1410) (resetting Tor).
- If still no luck, try the following commands and repeat the steps before:
```
sudo ifdown eth1
ip addr flush dev eth1
sudo ifup eth1
```
- If finally connected with the Internet, use entry 5 in the Update and Maintenance menu (updating the TorBox menu)
- After leaving the Main Menu, execute the following command: `sudo cp etc/rc.local /etc/rc.local`
- With the next restart, everything should work as intended.

closed 2025-07-07 by radio24