zotil/TorBox #339: Dual WiFi/Ethernet Clients

plzdontbanme, 2025-01-30:
I have a TorBox on a Raspberry Pi 5 set up and running per the instructions. I have it configured to use the on-board ethernet (eth0) for WAN.

When I connect to the WiFi, I am able to access the WebSSH (192.168.42.1) and am successfully able to reach websites (e.g. check.torproject.org).

When I plug in a USB Ethernet adapter and connect to that, I am able to access the WebSSH (192.168.43.1), but I cannot reach any websites. The logs show that Tor is connected properly, the WebSSH console says "Tor is working", but trying to reach any website (including check.torproject.org) fails with a timeout.

Am I missing something? Other than needing a USB Ethernet adapter, the instructions don't say anything about any other configuration changes needed to run dual WiFi/Ethernet clients with eth0 as the WAN port.

hal9000node, 2025-02-01:
I have also had considerably more success with the wireless abilities of this device and find the eth1 to be a bit hard to use.

radio24, 2025-02-02:
Could you please check the following entry in `torrc`:
```
DNSPort 127.0.0.1:9053
#DNSPort 192.168.42.1:9053
DNSPort 192.168.43.1:9053
#DNSPort 192.168.44.1:9053
```

Important is this line: `DNSPort 192.168.43.1:9053`

radio24, 2025-02-02:
Feedback suggests something may be wrong in uncommenting `DNSPort 192.168.43.1:9053` in `torrc`. Let's try to find the bug together, is there one.

1. Flash a new SD Card by using the latest image file: TorBox Image (about 1 GB): [v.0.5.4 (11.01.2025)](https://www.torbox.ch/data/torbox-20250111-v054.img.xz) – [SHA-256 values](https://www.torbox.ch/?page_id=1128). The compressed `.xz` file has to be decompressed to flash the `.img`.
2. Put the SD Card into the Raspberry Pi, connect it to an Ethernet cable to get the Internet and start it up.
3. Wait 1-2 Minutes after the system on the SD card is expanded and restarted again; connect to the AP called `TorBox054` as well as with a SSH client on `192.168.42.1`.
4. In the first start-up dialogue, select the default options and Internet source entry 1, already highlighted with an arrow.
5. TorBox is running, and the client reaches [check.torproject.org](check.torproject.org) via the tor network.
6. Shut down the TorBox via main menu entry 19.
7. Let the Ethernet cable with the Internet plugged in, plug an USB Ethernet adapter into an USB3 socket and connect the client with an ethernet cable as pictured [here](https://www.torbox.ch/?page_id=112#my-torbox-is-connected-with-the-internet-via-ethernet-im-using-onboard-wifi-for-my-client-devices-how-can-i-additionally-connect-a-cable-client) (don't forget deactivating wifi on the client and to take care that the Ethernet interface on the client is configured using DHCP).
8. Start the TorBox again; the client's Ethernet interface gets the IP `192.168.43.10` and connect with an SSH client on `192.168.43.1`.
9. In the main menu, select entry 5 (already highlighted with an arrow.) to reconfigure TorBox to work with the new connection type (the Internet source is still the same, but not the client - this has to be configured) --> tor does restart with the new configuration.
10. TorBox is running, and the client reaches [check.torproject.org](check.torproject.org) via the tor network --> all works as intended.

Checking `/etc/torrc` ... it looks like that:
```
DNSPort 127.0.0.1:9053
DNSPort 192.168.42.1:9053
DNSPort 192.168.43.1:9053
#DNSPort 192.168.44.1:9053
```

I'm sorry, but I don't understand why it isn't working with your configuration. Did you use the latest image? Did you rerun the main menu entry after changing from wifi to the Ethernet connection? This is necessary because you changed the way the client is connected. In the background, the Tor configuration and the tables must be adjusted to that change.

radio24, 2025-02-02:
Now, let's try to reach [http://archive.ubuntu.com/](http://archive.ubuntu.com/):
1. Go to the configuration sub-menu. By using menu entry 15, disable the HTTP plain text traffic block.
2. With a browser, connect to `http://archive.ubuntu.com` --> working!

plzdontbanme, 2025-02-06:
@radio24 I didn't do step 9 (reconfigure torbox for the new network), because that step isn't documented anywhere that I could find.

Does this mean only one can be used at a time? Or, after I do that step, can clients connect with both Ethernet and WiFi?

radio24, 2025-04-26:
You can only have one Internet source, but you can have clients on WiFi and Ethernet simultaneously. See also [here](https://www.torbox.ch/?page_id=112#my-torbox-is-connected-with-the-internet-via-ethernet-im-using-onboard-wifi-for-my-client-devices-how-can-i-additionally-connect-a-cable-client).

closed 2025-07-07 by radio24