zotil/TorBox #342: 🚫  Tor over VPN not working...

BeatrixIssac879, 2025-03-07:
Good day! Thank you very much for the wonderful product. We are very grateful to you.

I have a problem that I can't solve, I need to set up "Tor over VPN" I have a working connection to the VPN and I see "VPN is up & TOR working" but below in the "Where is the Internet?" tab I see "5 Ethernet cable (eth0) or TorBox on a cloud (tun1) <-- " based on this I understand that the traffic works through eth0, the question is what am I doing wrong? Why, when choosing 10 Over a VPN connection (tun0), it connects to the VPN but the traffic still goes through eth? How can I overcome this?

Notes,

I tried UDP, I tried TCP, the result is the same.

radio24, 2025-04-21:
Hey,
Thank you for reporting your issue. 
I suppose it's a bug that shows the "<--" beside the wrong menu entry. To ensure we understand what's going on, we will investigate, but we need some time to do so. We will respond to you here in this thread. Thank you for your patience.

radio24, 2025-05-07:
Ok, I looked into the problem.
For some reason, I missed adding a line, which I just did with commit 6b9af96 (Tor over VPN: fix).
Thanks again for letting me know.

closed 2025-07-07 by radio24