zotil/TorBox #369: Custom onion services option or more options like an irc server and client for those more used to irc than what tor chat secure provides

MCHAMSTERYT2, 2025-07-28:


MCHAMSTERYT2, 2025-07-28:
Forgot the message but added it in now so should be a good feature request

MCHAMSTERYT2, 2025-07-28:
Also sorta unrelated to this feature request but for those using a raspberry pi 3b (non-plus) would it be possible to implement a feature to use a second wifi adapter in access point mode since the 3b has only 2.4g built in wifi limited to about 100 mbps or under?

MCHAMSTERYT2, 2025-07-28:
Yeah since then the speeds will be only limited by usb 2.0 speeds and the adapters themself which even as slow as usb 2.0 is it's still much faster than the built in wifi

radio24, 2025-07-28:
Hmm, I have to look into it. It will be a long-term project. First, I will test whether a bridge interface could be a better alternative for Ethernet interfaces. This might lead to some more insights, which are also interesting with the handling of wlan interfaces.